aredn/aredn #372: Typo in PR 371

Orv, 2022-05-25:
IS: <li>blocked - latency: the link latency is to high to reliably pass data.</li>
SHOULD BE: <li>blocked - latency: the link latency is **too** high to reliably pass data.</li>

ab7pa, 2022-05-25:
I'm looking but just not seeing it... can you identify the line number?  THX

<a name=lqmstatus></a>
<ul>
<li>pending: LQM is collecting data and evaluating the link.</li>
<li>active: the link is viable and can be used.</li>
<li>idle: the link is usable and would be <em>active</em> but the node routing table does not yet have a route for sending traffic across the link.</li>
<li>blocked: the link is unusable and is blocked from use.</li>
<li>blocked - distance: the remote node is either too close or too distant, based on the LQM Min and Max Distance settings.</li>
<li>blocked - signal: the SNR on the link is too low to reliably pass data, based on the LQM Min SNR setting.</li>
<li>blocked - retries: the retransmission rate is too high to reliably pass data.</li>
<li>blocked - latency: the link latency is too high to reliably pass data.</li>
<li>blocked - dtd: LQM blocks the RF interface of any nodes to which a DtD link also exists.</li>
<li>blocked - dup: LQM blocks a link in cases when your node has an RF link to other nodes which themselves connect to each other via DtD. This can occur when there are multiple radios at a site using the same channel. The best remote node is chosen as the RF link for your node but the other possible RF connections are blocked as duplicates.</li>
<li>blocked - user: LQM blocks any node which you enter in the <em>User Blocked Nodes</em> field in the Advanced Configuration section.
<li>disconnected: the RF Neighbor is no longer online.</li>
</ul>

Orv, 2022-05-25:
It's line 83, but in your example, there it's been fixed.  This is what I'm seeing:

<li>blocked - latency: the link latency is **to** high to reliably pass data.</li>

ab7pa, 2022-05-25:
Here is line 83 as I'm seeing it.  I think in the future it would be easier to troubleshoot and fix things if you would consider making your comment directly on the PR rather than opening a separate issue.  In the PR you can comment on the exact line of code where the problem occurs.
[image]

Orv, 2022-05-27:
Roger Wilco

closed 2022-05-27 by Orv