aredn/aredn #376: LQM node block appears non-functional

wa3lab, 2022-05-28:
First off, if this is not the place to list issues with nightly builds, please tell me where to list them.  I have nightly build 1315-ba94a86 installed and had updated from build 1287.  1315 has resolved almost all the issues I saw in 1287 except for one.  I have the distance set to 9 miles, and all other parameters are set to default except I have entered 2 MAC addresses that I would like blocked.  One of the stations has a good SNR (19-20)is 17 miles away, and the Neighbor Status screen shows "blocked - user".   However the node with the good SNR continues to connect and shows up as a current neighbor.  It also registers on my mesh map as a green line.  This connection is problematic and has been found to create issues in the past.  My assumption is that If I block it specifically by MAC address it should be blocked regardless of the signal strength? Please let me know if I am correct.  Thank you for you efforts.
73,
Glenn WA3LAB

aanon4, 2022-05-28:
Apologies, but I think I need a little clarification to help:

1. You mention you blocked two MAC address, but are we only discussing one that is problematic? Is the other okay?
2. The problematic one is user-blocked, too far away (more than 9 miles), but the SNR is good, correct?
3. When you say the nodes "continues to connect and shows up as a current neighbor" ... what exactly do you mean? Is this the neighbor shown as "blocked" but is a neighbor listed somewhere else? If so, where exactly?

Two things which would help to clarify things:
1. Screen shots of both the mesh status page and the neighbor status page.
2. A support data dump.

Your final assertion about blocking specific MAC addresses is correct, so let's make sure that's behaving as intended.

ab7pa, 2022-05-29:
As is stated in the online documentation, "Nodes are blocked by their **Wifi** MAC address which must be all uppercase with the hex pairs separated by colons. **Whitespace** is not permitted in this list of MAC addresses."  Could the behavior Glenn is seeing result either from entering the wrong MAC address or from whitespace in the comma-separated list?

aanon4, 2022-05-29:
Possibly. A space after the comma between two MACs would do it for example; but without more information it's impossible to know.

wa3lab, 2022-05-29:
Here is the setup.  The neighbor status is "blocked-user"  when it is 
failing.  The 3rd MAC address is a test node  where I am trying to 
reproduce the issue.  I'll you know if I do.

73,

Glenn WA3LAB

wa3lab, 2022-05-30:
Tim,

     I inadvertently replied incorrectly to your first email.  I sent a 
second email where I replied to AB7PA, and it shows up on Github, but 
without the screenshot.  My sent email shows the screenshot but it 
doesn't show on Github.  Some type of cockpit error I assume.  Just for 
your information, the node that I had blocked now has been removed from 
the current neighbor section and is actually being blocked.

Glenn WA3LAB

aanon4, 2022-05-30:
Glen,

No worries, and thanks for the update. Can we close this issue?

Tim

dman776, 2022-05-30:
Are we stripping white space on the block list?

aanon4, 2022-05-30:
No. The block list is very particular about the format which is valid, and it's been documented exactly the format to use.

wa3lab, 2022-05-30:
Tim,

     Yes you can close the issue.  I'll keep an eye out for it if it 
happens again.  I am going to put a screen shot below.  Let's see if it 
makes it.  Is is a shot of the Link Quality parameters.

73,

Glenn

wa3lab, 2022-05-30:
I have been responding via my email client.  Responding via Github works for screen shots. 
Glenn
Here are the parameters.
[image]

aanon4, 2022-05-31:
Thanks for this. Is it possible for you to post the information from my original reply?

wa3lab, 2022-05-31:
Of course I forgot the support data.  
Glenn WA3LAB
[supportdata-WA3LAB-2G15-2-202205282016.zip](https://github.com/aredn/aredn/files/8805795/supportdata-WA3LAB-2G15-2-202205282016.zip)

aanon4, 2022-05-31:
Thanks for this - super helpful.
How comfortable are you with replacing a single file on this node by hand to run a test?

aanon4, 2022-06-01:
So, I've made some changes which will be in the new build (I don't know the exact number, but it will be 1326 or higher). If you could possibly try this I would be very grateful. @wa3lab

wa3lab, 2022-06-01:
Ok, I'll give it a try.  I'm currently not home, but I will get to it later today or tomorrow.   I'd also like to discuss possibly getting more involved in the firmware development.  I have a background in embedded controls/servos, but no experience with Lua.  Thanks for your continued support. 
Glenn WA3LAB

wa3lab, 2022-06-02:
I have loaded nightly build 1332, and all appears to be working as advertised.  It took about 10 minutes to fill in all the pending nodes.  Looking good! Glenn
[image]
[image]

closed 2022-06-07 by aanon4