aredn/aredn #377: Channel dropdown missing / Ubiquity Rocket M2 Titanium after flash upgrade v 3.22.1

kg5ipa, 2022-05-29:
I upgraded my Rocket M2 Titanium to v. 3.22.1 and now the dropdown to choose RF channel is missing. 
Attached are the screenshots.
Now my antenna is not operable. 
[image]
[image]
[image]

ab7pa, 2022-06-03:
@kg5ipa  if you have command line access to your node, could you post the contents of the /etc/board.json file here?  THX

dman776, 2022-07-13:
closing no response