aredn/aredn #381: DHCP Server Weirdness

Orv, 2022-06-04:
We replaced a PTZ camera at a local node site about 4 months ago with a new one, same make & model (Sunba 405DX).  About two months ago it disappeared off the network.  Power-cycling it didn’t resolve the issue.  The Ethernet cable, POE injector and 48V power supply all tested good.

We pulled it down and I brought it back to my shack to investigate.  I connected it to my hAP via a POE injector.  It powered up, did a normal ‘patrol’ (cycling the camera lens around), indicating that it was alive.   But my hAP didn’t register an Ethernet address.  These are the same symptoms it showed at the network site.  There the DHCP server is running on a Rocket M5, but it’s the same code.  All nodes in the area, including my shack, are running some recent nightly build.

My hAP is configured for 13 Host Direct and has been that way for years.
It’s IP  Address is 10.95.109.97 and it issues IP addresses
from 10.95.109.98 to  10.95.109.110.  Netmask is 255.255.255.240

Using Sunba’s Device Manager, I found it at 10.95.109.104, yet the hAP doesn’t record it as having been issued, thus there’s no route to it, leaving it non-functional.

[image]

[image]

Several months ago I was loaned a slightly newer Sunba camera, the 405DX ECO.   I had similar issues with it powering up successfully but not appearing in the list of Current DHCP Leases.  Using the Device Manager, I found it at 10.95.109.35, which isn’t even in the range of addresses to be issued.  I’ve since returned that camera to the owner.

I’ve not seen this problem with other camera brands I’ve used (Reolink, Wyse, Hikivision, and some Chinese no-name).

Nor do I see it with a newer model Sunba 625, with IP addressed leased from a node running 3.22.1.0 production release code.

Here's the support data file, if that's any help.  Renamed as .zip to allow upload.

[supportdata-W6BI-Shack-Node-202206041111.zip](https://github.com/aredn/aredn/files/8838392/supportdata-W6BI-Shack-Node-202206041111.zip)

Any ideas?

aanon4, 2022-06-04:
So this happens when one side of the DHCP handshake doesn't complete all the steps, usually the server doesn't hear the acceptance of the offer. Can you confirm one thing and try another thing? Confirm there are no other active dhcp servers? And can you try plugging the camera into a separate setup with a different dhcp server?

aanon4, 2022-06-04:
Could you resend the support data, but make sure to capture it about 1 minute after first powering up the camera. I don't see it even attempting to get a DHCP address, but perhaps it tried earlier and the log file doesn't capture it.

Orv, 2022-06-04:
Confirmed no other DHCP servers (it's only the hAP in the shack).
here's the support data immediately after the camera powers up.

[supportdata-W6BI-Shack-Node-202206041411.zip](https://github.com/aredn/aredn/files/8838815/supportdata-W6BI-Shack-Node-202206041411.zip)

aanon4, 2022-06-04:
I'm going to assume that by "immediately after the camera powers up" you really meant "capture it about 1 minutes after first powering up the camera" ... yes? I don't see any requests in the log for this MAC or anything related to the camera. I'd hoped to see something.

In the "DeviceManage" tool, if you select the device, what other information do you get?

Finally, do you fancy running tcpdump on your network to just capture the DHCP traffic as you power up the camera?

ae6xe, 2022-06-05:
Do you by chance have any daisy chain AREDN nodes via the secondary port and POE pass though?    In this situation, it can be confusing as this setup daisy chains the devices as if there is only 1 big switch with 4 ports.  Thus, when the 2nd node powers up, all it's LAN devices will obtain an IP address from the 1st node.  The 1st node's DHCP server is live first, so the 2nd node's DHCP will detect that and not go live.   Thus all LAN devices on both nodes will receive an IP address from the 1st node.

Orv, 2022-06-05:
Good thought, Joe, but no.   My hAP goes to a ToughSwitch at the base of the tower in the back yard.  From there the switch drives  2 GHz and 5 GHz nodes.   The 5 GHz node has its DHCP server enabled, but it has its own pool of IP addresses separate from those of the hAP, and they don't appear to be associated with this issue.

ae6xe, 2022-06-05:
Other possible use cases...   Maybe in a past deployment, the ipCam had a fallback static IP address from another node.   It can be configured DHCP, but when there's no response, it defaults back to a given IP address.

Orv, 2022-06-07:
OK, this camera is definitely not behaving normally.  I'm going to write it off as broken and we've ordered a replacement.

Orv, 2022-06-08:
I'm closing this - I think the cameras damaged.

closed 2022-06-08 by dman776