aredn/aredn #385: Incorrect Channel Selection Wi-Fi AP Mode on 5GHz devices N-1348

k1ky, 2022-06-08:
Wi-Fi AP Model Channel Selection shows Channels 1-11 (2GHz) instead of 5.0 GHz Channels on various 5GHz Single-Band Models.  Verified this issue exists on UBNT Nanostation M5, M5XW, Rocket M5, PowerBeam M5-400-ISO and Nanobridge M5.  I suspect this is uncharted territory and may exist on prior Firmware versions.  I have attached a screen shot and various Support files for several models
[supportdata-K1KY-VARIOUS-2022-202206071729.zip](https://github.com/aredn/aredn/files/8858222/supportdata-K1KY-VARIOUS-2022-202206071729.zip)
.

aanon4, 2022-06-08:
Let me take a look. Unfortunately I don't have a 5GHz only device on my desk.

k1ky, 2022-06-08:
Same issue verified with Mikrotik LHG5 as well.

aanon4, 2022-06-08:
Definitely a "dump_perl" lua bug. Will fix it now.

aanon4, 2022-06-08:
This should fix if. I ran a test on a local NSM5 node to make sure the correct UI now shows up.

k1ky, 2022-06-08:
Excellent!  Next up will be to test the WI-Fi as WAN on the 5GHz models.  I figured we would get past this issue first!

k1ky, 2022-06-08:
N-1357 fixes this issue.  Thanks for the quick response!

closed 2022-06-08 by k1ky