aredn/aredn #387: how to setup a meshmap by self？

ggsnpdsn, 2022-06-08:


ae6xe, 2022-06-08:
@ggsnpdsn, please clarify.   Is this question about the difference between AREDN firmware and, e.g. AirOS firmware?    AirOS firmware is required by the part 15 rules to ensure the device stays within part 15 rules (to be able to market and sell the device).    AREDN devices are operating under part 97 ham radio rules, and a similar rule does not exist (and we are not marketing anything).   Common with ham radio equipment, the licensed operator is responsible to consider the compliance of operating the device.

ggsnpdsn, 2022-06-08:
I don't want to know about the difference between the aredn firmware and the airs firmware. Technically, I want to know whether it is possible to expand the spectrum in a specific space by modifying the backport, IW, HostAP, etc. I'm sure that other wireless devices will not be disturbed in this specific space

ae6xe, 2022-06-08:
The AREDN ath9k backport changes are publicly visible at the link below -- an example of expanding the spectrum or channels outside of part 15 allocations.  
  
https://github.com/aredn/aredn/blob/main/patches/701-extended-spectrum.patch

dman776, 2022-06-08:
What does this have to do with meshmap?  Besides, that software was not build by AREDN, it is Eric's.

ggsnpdsn, 2022-06-08:
> The AREDN ath9k backport changes are publicly visible at the link below -- an example of expanding the spectrum or channels outside of part 15 allocations.
> 
> https://github.com/aredn/aredn/blob/main/patches/701-extended-spectrum.patch

Is the ath10k modification method the same?

ggsnpdsn, 2022-06-08:
> What does this have to do with meshmap? Besides, that software was not build by AREDN, it is Eric's.

I should have revised the title. I don't know why I didn't revise it

ae6xe, 2022-06-08:
"Is the ath10k modification method the same?"

Generally, yes.   What are you looking to do?   

Take a look at:

1) https://github.com/aredn/aredn/blob/main/patches/702-enable-country-hx.patch
2) see:  https://wireless.wiki.kernel.org/en/users/drivers/ath10k

specifically, "applying ath9k regulatory domain hack patch from OpenWRT causes firmware crash (reason: regulatory hint function is never called and ath10k never sends scan channel list to the firmware which in turn causes firmware to crash on scan)".  May still be an issue, not clear if anyone in the opensource community has implemented this to date.

closed 2022-07-13 by dman776