aredn/aredn #389: Expected internet traffic when operating tunnel - 800+MB/day?

nathantcampbell, 2022-06-10:
Hi,

We have a couple nodes that are remote and operating via a tunnel through a cellular hotspot until we get RF links out to them. We've found that even with no device-driven traffic, the HAP consumes about 850 MB / day or 3 MB per minute. It is our suspicion this is related to OLSR updates? We're on the Willamette Valley Mesh network which hovers around 270 nodes.

Is it likely this continuous traffic is OLSR updates? Is there any way to manually mitigate how often this occurs on these remote devices to decrease the data usage?

Controlled test case is a HAP with no devices, tunneled into the network (single connection so no traffic should be routed through the node).
Downloading 3MB/5min and uploading 200KB/5min.

[image]

Nathan, AI7NC

VA2XJM, 2022-06-10:
Our group have an LTE backup link at one site. To mitigate this, we enable the tunnel link to that site ONLY when it is needed and for the shorter amount of time possible due to cost and bandwidth limitation. In our case, the remote LTE site tunnel is always active and we can enable/disable the link from the main tunnel "server" device whenever it is needed.

dman776, 2022-06-10:
Do you have mesh gateway enabled on this node?  (Ie.   Allow nodes on the mesh to access my wan connection?)

Thanks,
Darryl

nathantcampbell, 2022-06-10:
Negative, plus there are no devices connected to the HAP. 

[image]

aanon4, 2022-06-10:
Back of the envelope calculations, OLSR will send ~ 2 MB of traffic across a link per node on the network per day. It's the way the protocol works I'm afraid. So with 270 nodes my math puts it lower than you're observing, but clearly in the right ballpark.

nathantcampbell, 2022-06-11:
Gotcha, is the traffic pushed or pulled based on parameters set on the node? Just thinking around if there's an option to decrease the update schedule if it's a pull.

aanon4, 2022-06-11:
You can reduce the traffic, but as it's pushed from each node, you'd have to do that at every node.

nathantcampbell, 2022-06-11:
Ok well I think that's plenty of evidence to pin this on OLSR updates, bummer it's pushed so the only way to control it for these sites is to have the link disabled. Thanks for the help everyone, we can close the issue.

closed 2022-06-11 by nathantcampbell