aredn/aredn #392: CONFIG_VTUN_SSL=Y ?

mathisono, 2022-06-12:
The VPN that connect "Tunnels" is currently configured in a manner, that's considered a week mode of authentication! I Concur with this statement.  The Clear text for password? over the internet? 

Q: Should it be considered for a road map item amendment. or am I wrong?  Do I just need to =y on the server and then it will enable? https://github.com/aredn/aredn/blob/1ceb7b21400e1d57cc01c6701418a9987d8fb5fc/configs/common.config#L116

https://github.com/aredn/aredn/blob/1ceb7b21400e1d57cc01c6701418a9987d8fb5fc/configs/common.config

73 Mathison

aanon4, 2022-06-13:
Just to clarify the requirements - the needs it for secure authentication between tunnel server and client, but not data encryption beyond that?

aanon4, 2022-06-13:
A very preliminary look at the vtund code, and the AREDN version has BlowFish (symmetric encryption) and MD5 (secure hash) built in now. These appear to be used for client/server authentication only. You need the SSL library if you want to encrypt your data (not your authentication). This all appears to be on by default.

So .. wouldn't swear to it without doing a proper network capture on a tunnel setup .. but to me it looks like authentication is already secure, but data isn't.

Enough to make CAL happy? I can do a full dump of the tunnel setup handshake to make sure and/or document what it actually does (you'd need to send me a tunnel server I can talk to - I don't have any setup myself).

aanon4, 2022-06-13:
The authentication protocol works as follow:

Alice (the server) creates a large random sequence of bytes and sends them to Bob (the client).
Bob encrypts the sequence using a hash of the shared password.
Bob sends the encrypted sequence back to Alice.
Alice decrypts the sequence using the same hash of the shared password. If the decrypted sequence matches what Alice sent, the channel is authenticated. No plain text passwords are ever exchanged.

mathisono, 2022-06-13:
you beat me to a response!  Im not asking but if I wanted to tunnel encrypted. Lets complicate the matter in a number of ways...! The server establishes with the client and defines the connection is needing to be encrypted, although it falls on to the client to ensure that its not going to establish an encrypted tunnel out over HAM spectrum and only, go out over its WAN.  A hard coded condition?  I'm just glad to confirm that no plain text is in the mix. Tim you have clearly documented  the "authentication protocol work follow" This helps!

ae6xe, 2022-06-13:
Historically data encryption was turned off at one point.   The tunnel processes were consuming ~500k on each end per link, and we had reduced the RAM foot print by a non-trivial amount.   I don't recall by how much, however.  Part of the issue, as I recall, was the implementation used a heavy library for encryption, and we likely could still do encryption (over the internet, not RF) with a lighter library option.   Time, interest, priority, etc. to do.

mathisono, 2022-06-14:
Secure tunnel authentication for server and client! Great to confirm!!! I don't intend to insist that the Tunnels should have link encryption. The local security team has advised me of a new policy that's no longer a daft to policy!  https://security.berkeley.edu/MSSND/use-authentication-guidelines.