aredn/aredn #395: latest ARDEN MESH firmware update vs Ubiquity PBE M5 300

kk4ib, 2022-06-13:
Just flashed the PowerBeam M5 300 with 3.22.1.0-ar71xx-generic-ubnt-loco-m-xw-squashfs-sysupgrade.bin;  and now the Status page won't populate; all other pages work fine and the PBE joined the network via router port and via RF.
[supportdata-KK4IB-PBE-M5-202206131806.zip](https://github.com/aredn/aredn/files/8893916/supportdata-KK4IB-PBE-M5-202206131806.zip)
[image]

dman776, 2022-06-13:
Try it with the latest production release 3.22.6.0

aanon4, 2022-06-13:
https://github.com/aredn/aredn/pull/396
This fixes unknown radio models being displayed. I'll leave the actually support of this model to @ae6xe

ae6xe, 2022-06-13:
Same comment - Ref:  https://github.com/aredn/documentation/issues/172#issuecomment-1154389350

kk4ib, 2022-06-17:
This issue was FIXED with Nightly Build 3.22.6.1; the PBE M5 300 is working properly.

closed 2022-06-17 by kk4ib

kk4ib, 2022-10-11:
That’s what I did to cause the issue. Using 3.22.1.0 it works fine.

It actually works fine with "22.6” also, just won’t populate the Status page.

sparks