aredn/aredn #401: New Mikrotik HAP AC3 - Model Support Request

k1ky, 2022-06-17:
Request to add Mikrotik HAP AC3 to AREDN.  It is supported by OpenWRT.
ipq40xx: add support for MikroTik hAP ac3
This adds support for the MikroTik RouterBOARD RBD53iG-5HacD2HnD (hAP ac³), a indoor dual band, dual-radio 802.11ac wireless AP with external omnidirectional antennae, USB port, five 10/100/1000 Mbps Ethernet ports and PoE passthrough.
See https://mikrotik.com/product/hap_ac3 for more info.

ae6xe, 2022-06-17:
The HAP AC3 is on the list along with many 802.11ac devices.    I have very early images on this device now.  However, there remains work to get the 802.11ac wireless working as the code is proprietary and there is  a need to work with 3rd parties that have NDAs and rights to make changes.   So all 802.11ac devices are in this category.

k1ky, 2022-06-17:
Shall I change the title to broaden the description or is there another issue already covering this??

k1ky, 2023-01-15:
It appears that we have some model support in the nightlies - do we keep this issue open until Production release?

aanon4, 2023-02-08:
I think we'll close this now as it's stable in nightly. Helps us track what's pending before the next release.

closed 2023-02-08 by aanon4