aredn/aredn #404: LQM Disconnects Mesh Node from Network

tsdxmad, 2022-06-17:
After installing the latest firmware version 3.22.6.0 on an Ubiquiti M5 XW, I enabled LQM without making any changes. The link quality was not the greatest to other nodes, but worked with a SNR of 15 or 16 to one of our backhaul nodes. After about 8 min, LQM blocked all signals and knocked the node off the network. I did manage to get a screen shot of "Neighbor Status" showing this, just before getting kicked to the curb. Please see below. It seems the node should leave the best connection active and not block all regardless of settings.
[image]

.

Orv, 2022-06-18:
Drop the default SNR to 8 or less (undder Basic Setup) and they'll come back. Then encourage them to improve their links; 15 dB is about the minimum to have decent throughput and not have too many retransmits.

aanon4, 2022-06-27:
What makes a good 'failsafe' behavior remains unclear. At the moment, when a node is first setup and it cannot find a viable link, it reduces its requirements until at least one RF link is valid. Here, a link was valid originally, but became invalid over time (because radio) cutting off the node. Not great. However, what is a good recovery strategy? Making sure a single RF connection works doesn't necessarily get the node back not he mesh ... maybe that one connection is just to a local neighbor which doesn't get you much (which is also a limitation of the current strategy).  Dropping the requirements to allow all RF nodes to work would result in poor performance for the node, and impact general mesh performance.

Other options might include having much lower requirements for nodes which are upgraded to LQM (rather than newly installed nodes which might have more idea requirements) to reduce the chances of cutting off existing nodes. But this also reduces potential performance improvements.

I'm open to suggestions and thoughts on this.

closed 2022-07-22 by dman776