aredn/aredn #405: DHCP reservations not being honored

k6ccc, 2022-06-21:
Posted here at the request of Andre after posting this on the AREDN forum.

Here is my original post:
This has been a somewhat re-occuring issue for quite some while, but I may have noticed something consistent.
Off and on for several years, I have had devices connected to LAN ports of either of my hAP nodes that have DHCP reservations, but the reservation was not honored, and some other address was assigned.  Have not seen that for a while, but after a bunch of re-arranging here, I noticed the problem showing up on both hAPs with multiple devices.  This morning I noticed a consistency.  Every device that was getting a wrong address, under DHCP leases in the hAP was showing the hostname as being the IP address that it should have had based on a DHCP reservation and the Current DHCP Lease showing the assigned address and MAC.  Unfortunately I did not screen capture this condition.  The part that was consistent was that EVERY device that had the Do Not Propagate checkbox selected was in this incorrect state.  I unselected the DNP box on one device and saved it in the hAP and then on the device (a router in the first case) I released and renewed the DHCP client and promptly got the correct address based on the DHCP reservation.  I repeated the test on several other devices, with the same results.  As soon as I unselected DNP and renewed the lease, the hAP assigned the correct address based on the DHCP reservation.  Re-selecting the DNP checkbox and renewing the lease did not immediately result in an incorrect IP address being assigned.

Both of my hAPs are currently running 3.22.6.0

And here is a followup a couple hours later:
It's a couple hour later and on both hAPs, one of the devices that have a DHCP reservation with Do Not Poroagate checked has changed the hostname in the DHCP leases column to the IP address.  When I did my earlier post, the host names were proper after unchecking DNP.  So far, the IP has not changed.  This time I took screen captures - before and after:
[image]

I downloaded a support file after the second post, but not allowed to upload here.  I have posted the file to my website so you should be able to download it from there.  If there are problems with that, I can E-Mail it to someone.
http://k6ccc.org/aredn/supportdata-K6CCC-hAP-Portable-202206201409.tgz

aanon4, 2022-06-21:
This is fantastic information. I have been chasing this bug on an off but had never put my finger on exactly when it occurred - this observation about "do not propagate" is key - thanks.

aanon4, 2022-06-21:
I *think* this has been an issue as long as I've been using AREDN (so about a year) but I wondered if that's also your observation?

aanon4, 2022-06-21:
It looks like dnsmasq (the code handling DHCP) doesn't handle # comments on the ends of lines in /etc/ethers. Not sure of the best fix yet as various pieces of AREDN code rely on this being there.

aanon4, 2022-06-21:
Fix https://github.com/aredn/aredn/pull/406

k6ccc, 2022-06-21:
Yes, it has been around for a while (I think).  For background, I'm the guy that requested the Do Not Propagate feature several years ago, so I guess it's all my fault :)
Glad I was able to provide a useful key!

k6ccc, 2022-06-22:
Appears to be fixed in Nightly 1421.
Thanks

k6ccc, 2022-06-23:
Nightly 1421 fixes this.

closed 2022-06-23 by k6ccc