aredn/aredn #413: Sort function not working properly in wifi scan

Orv, 2022-06-26:
in nightly build 1449, the sort on channel function is broken; it only sorts AREDN nodes:

[image]

aanon4, 2022-06-26:
Can you provide a little more info? I don't see an arrow on any column indicating a "sort" selection, and prodding my own scan page it all seems okay - so what am I missing?

Orv, 2022-06-26:
All the columns will sort; just click on the column heading.

aanon4, 2022-06-26:
So .... what is the bug exactly? The initial sorting is sorted by signal (as it always was) and you say sorting by column heading works too. So I'm unclear what is broken.

Orv, 2022-06-26:
Sort on channel is NOT working.

Orv, 2022-06-26:
Hrm - can't reproduce it.  Closing ticket until I can.

closed 2022-06-26 by Orv