aredn/aredn #415: STXsq Lite2 no neighbors with LQM enabled

AJ6GZ, 2022-06-26:
With LQM enabled, STXsq Lite2 does not stabilize its links with other nodes. During this time LQ/NLQ is 100/0% with all RF neighbors, occasionally rising but falling back to zero. I noticed these models do not work with a distance Auto (0) setting while LQM is OFF and it exhibits the same behavor as above. Works fine with LQM off and non-zero distance.  Tested several times and it did come up and stabilize in about 20 minutes once, other times not at all. Tried with two different SXTsq's and Nano's in the mix. 
I can put a spare on the SoCal network if someone wants to poke around.

MikroTik SXTsq Lite2 RBSXTsq2nD
Build 1449 (and others prior)
Ch -2/10Mhz
[supportdata-AJ6GZ-Portable1-SXT2-202206242008.zip](https://github.com/aredn/aredn/files/8987632/supportdata-AJ6GZ-Portable1-SXT2-202206242008.zip)

aanon4, 2022-06-26:
Could you post a screenshot of the Neighbor status page with LQM enabled (after the status's no longer say 'pending')? Thanks.

aanon4, 2022-06-26:
This node doesn't have its location set?

ab7pa, 2022-06-26:
My Mikrotik SXTsq Lite 2 nodes link well with all other manufacturer devices, at least on channel -2 (5 MHz width) which is what our net is running.  They have worked with LQM enabled or disabled -- both modes work fine, and auto-distance (zero) also works.

AJ6GZ, 2022-06-26:
Been up for an hour... location not yet set
[image]

Location set, after 20min uptime. Still no good connection.
[image]

(That one neighbor is expected to the blocked)

aanon4, 2022-06-26:
Given the SNRs I assume these other nodes as essentially in the same place? What does their neighbor info look like?

AJ6GZ, 2022-06-26:
Yes normally two nodes on-site, one down the street.  The 3rd one being tested is with a portable setup powered on.  However the other fixed SXT2 does the same thing.

Basically the opposite LQ/NLQ at any given time... happened to be 10% at this moment... as seen from the Nano:
[image]

aanon4, 2022-06-26:
The low LQ/NLQ values indicate a very poor connection between the two nodes (lots of packet loss) which I'm assuming is because of the two nodes inability to work out how far away they are from each other. In this circumstance, the node just reverts to "auto distancing" mode just as if LQM is off ... but you note that auto distancing with LQM off doesn't work correctly like this either.

Why auto distancing doesn't work on this node I don't know. If you were to downgrade the portable (the original) node to 22.1, does that fair any better? I would expect it to behave the same as LQM disabled, auto-distancing enabled but it would be useful to validate the fact.

AJ6GZ, 2022-06-27:
Test 1) Tested with 3.22.1.0 and got the same non-working result with Auto enabled.  Set a distance and it comes right up. Also reduced power on all nodes to 10dBm with same results both ways so not overload.

Test 2) Turned off all nodes except the one SXT still on 3.22.1.0. Tried to connect with node down the street (all of 200yds away). It worked with a distance set. It worked 5 of out 6 times with Auto and after upgrading with LQM set. I think the one time it angered the other side... I powered off for two minutes and it came up. The remote side is a NSM2 on 1279, no LQM.

I have a feeling Auto or LQM just can't handle a low physical distance scenario on this particular hardware. All are within a 75ft triangle here on-site. Works fine on the Nano XW.

aanon4, 2022-06-27:
So, in summary, auto distancing (the Linux version - not the LQM version) appears to be broken on this device specifically. As the kernel code is no different I don't know why this should be, and don't have any good way to work it out.

But as setting the distance by hand can work, I'm thinking we should add an advanced setting to allow an "if all else fails use this distance" setting which would at least circumvent this problem.  Thoughts?

ae6xe, 2022-06-27:
Auto distance is horrible at these short distances, less than ~3km.   This is expected behavior, and on-node help info and Read-the-docs recommends to use a static setting at short distances.

AJ6GZ, 2022-06-29:
After 1457...

Test 1) Node WITH location set: 
auto_distance (0) 100%/50% or less as expected.
auto_distance (1), 100%/100%

Test 2) Node WITHOUT location (the portable node):
auto_distance (0) no connection as expected.
auto_distance (1), 100%/100%

LQM seems to be doing its business blocking a weak signal.

I think that did it!

closed 2022-06-30 by AJ6GZ