aredn/aredn #419: NAT Mode DHCP range fields need data validation

ab7pa, 2022-06-29:
Forum post reports "Bad Gateway" messages in certain cases when using NAT Mode.
Example: [https://www.arednmesh.org/content/navigating-cgi-binsetup-gives-502-bad-gateway-status-code](https://www.arednmesh.org/content/navigating-cgi-binsetup-gives-502-bad-gateway-status-code)

I looked at the support dump this operator posted on the forum and I see in /etc/config.mesh/_setup that "dmz_mode = 0" and that "dmz_dhcp_end = " (empty value). On my test node when I clear the _DHCP End_ value on the _Setup_ page, press _Save Changes_, then the _Bad Gateway_ message appears. Some data validation could prevent empty values and force numeric values in those fields when in NAT Mode.

aanon4, 2022-07-13:
So looking at /etc/config.mesh/_setup I see the following values:

dmz_dhcp_end = 
dmz_dhcp_limit = -1

Clearly _end is blank which is bad, but _limit is -1 which is also bad. I'm not sure how this file got corrupted (I assume by the perl code at some point in the past, and it's much more blasé about empty strings being the value 0) but the user could edit this by hand to fix it. I dont think the current lua code can produce this problem.

aanon4, 2022-07-13:
https://github.com/aredn/aredn/pull/430