aredn/aredn #425: Spectral view not working on Rocket M3 with nightly 1479

k6ccc, 2022-07-02:
Spectral display blank on Rocket M3 XM
BTW, looks great on all the other nodes I have updated.
[image]

k6ccc, 2022-07-02:
Note that all three of my M3 Rockets are the same way.  M5 Rockets (both XM and XW) are fine.

k6ccc, 2022-07-02:
One more thing, I have a support data file, but can't upload that file type here.
Posted it to:  http://k6ccc.org/AREDN/supportdata-K6CCC-RocM3-1-1-202207021040.tgz

k1ky, 2022-07-03:
@k6ccc to upload .tgz file, I believe I either rename it or zip it to a .z extension.  Sure would be nice if this platform supported a .tgz extension.

k1ky, 2022-07-03:
I also get a blank spectral screen display with Nanostation M5 (XM) and Nanobridge M5.  No registration marks or spectrum, just a blank box as you depicted above.

ae6xe, 2022-07-04:
support dump shows this error on the K6CCC-RocM3-1-1 device: 

Sat Jul  2 10:25:34 2022 daemon.err uhttpd[1006]: /usr/bin/lua: /www/cgi-bin/scan:353: attempt to get length of local 'bucket' (a nil value)
Sat Jul  2 10:25:34 2022 daemon.err uhttpd[1006]: stack traceback:
Sat Jul  2 10:25:34 2022 daemon.err uhttpd[1006]: 	/www/cgi-bin/scan:353: in main chunk
Sat Jul  2 10:25:34 2022 daemon.err uhttpd[1006]: 	[C]: ?

k1ky, 2022-07-05:
I have another Nanobridge M5 with only 3 active connections that does have a functional spectral display.

aanon4, 2022-07-12:
This should be fixed in https://github.com/aredn/aredn/pull/429

k1ky, 2022-07-18:
All of my Nanobridges are functioning now with Nightly 1510.  No display issues with any devices known at this time.  Still wondering if the active frequencies are being displayed in the proper places?

closed 2022-07-21 by aanon4