aredn/aredn #428: DtD links not shown in Neighbor Status Occasionally - NB 1479

Orv, 2022-07-09:
I noticed that on a few nodes, the Neighbor Status omits the DtD links that are shown in the Mesh Status.  It doesn't seem to be band related; I've seen it on both 2 GHz and 3 GHz so far.  Here's an example:

[image]

[image]

upport data file from this particular node is attached (.tgz manually renamed to .gz)

[supportdata-W6BI-VC-SIMINORTH-3G-202207091206.gz](https://github.com/aredn/aredn/files/9077743/supportdata-W6BI-VC-SIMINORTH-3G-202207091206.gz)

aanon4, 2022-07-12:
If you have a node like this, can you attach the output from the following command:

Orv, 2022-07-13:
Here you go:

root@W6BI-VC-SIMINORTH-3G:~# cat /proc/net/arp
IP address       HW type     Flags       HW address            Mask     Device
10.106.154.60    0x1         0x2         68:72:51:6a:9a:3c     *        wlan0
10.111.33.194    0x1         0x2         14:cc:20:6f:21:c2     *        br-dtdlink
10.112.167.69    0x1         0x0         00:00:00:00:00:00     *        br-dtdlink
10.246.107.9     0x1         0x2         24:a4:3c:f6:6b:09     *        wlan0
10.65.50.70      0x1         0x0         00:00:00:00:00:00     *        br-dtdlink
10.15.20.232     0x1         0x2         44:d9:e7:0f:14:e8     *        br-dtdlink
10.246.107.9     0x1         0x0         00:00:00:00:00:00     *        br-dtdlink
10.112.167.65    0x1         0x0         00:00:00:00:00:00     *        br-dtdlink

aanon4, 2022-07-13:
https://github.com/aredn/aredn/pull/431

Orv, 2022-07-14:
Nightly build 1503 resolves this issue

closed 2022-07-14 by Orv