aredn/aredn #436: XSS is possible via the "service name" form field

decaffeinatedio, 2022-07-18:
Thanks for creating this great project!

**Issue**

While it appears that other input and display may be sanitized, some dangerous strings into the "Advertised Services" name, e.g. `serv1_name`, is not. This allows for cross-site scripting (XSS) attacks.  

**Steps to Reproduce**

Entering `<script>` tags generates an error, but this can be bypassed by providing a different tag. For example, entering: `x<img srx=x onerror=console.log(1)>` will successfully save. When viewed on the attacker's machine or on an end user's machine, it will display a broken image (at least on Firefox) and log "1" in the dev tools console. 

[image]
[image]
[image]

**Impact**
As this shows up on unsuspecting end user `*/mesh` pages, an attacker could deface or change end user's experience. I haven't tested this fully yet, but I also suspect that if the victim has already authenticated, the attacker could trigger configuration changes on their behalf:

```
let response = await fetch('http://localnode.local.mesh/cgi-bin/setup')
let reader = response.body.getReader()
let buffer = await reader.read()
let result = new TextDecoder().decode(buffer.value)

console.log(result)
```

aanon4, 2022-07-18:
Would it be sufficient to prevent < and > characters being used as part of a service name?

aanon4, 2022-07-18:
@decaffeinatedio I can't add you as a reviewer I'm afraid, but please take a look.

decaffeinatedio, 2022-07-18:
> Would it be sufficient to prevent < and > characters being used as part of a service name?

In my experience that should do it outside of some very unusual edge cases. I'll take a look at the PR now!

decaffeinatedio, 2022-07-18:
Resolved by https://github.com/aredn/aredn/pull/437