aredn/aredn #438: NB 1520 - Mesh Status Normal, Neighbor Status Blank

Orv, 2022-07-19:
[image]
[image]
[supportdata-K6PVR-VC-Sulphur-to-Oxnard-5G-202207191256.gz](https://github.com/aredn/aredn/files/9144277/supportdata-K6PVR-VC-Sulphur-to-Oxnard-5G-202207191256.gz)

First NB 1520 I've loaded.  Will look at others soon.

Orv, 2022-07-19:
Works properly on second node (different hardware).

aanon4, 2022-07-19:
Odd. No errors that I can see in the logs files. If you reboot does it "fix" things?

Orv, 2022-07-19:
I know what went wrong; rebooting it did in fact fix the issue.  Disregard!

closed 2022-07-19 by Orv