aredn/aredn #447: Invalid tactical name causes Bad Gateway error

pmilazzo, 2022-07-21:
Adding an underscore (which is syntactically invalid in a host name) to the tactical name of a node and then pressing Save Changes results in a Bad Gateway response. Observed in nightly 1534-02ec015 on an AR300M16.

Support data taken from a second browser tab immediately after seeing the error:
[supportdata-W3DRA-27-208-194-202207212056.gz](https://github.com/aredn/aredn/files/9162786/supportdata-W3DRA-27-208-194-202207212056.gz)
.

aanon4, 2022-07-21:
Any chance you could rerun this test and provide the supportdata again? Looks like some driver it the kernel died during this test and the log file seems truncated (an unrelated openwrt issue).

pmilazzo, 2022-07-21:
Is this the log section you're looking for?

Thu Jul 21 21:33:06 2022 daemon.err uhttpd[946]: /usr/bin/lua: 
/www/cgi-bin/setup:656: attempt to index global 'node' (a nil value)
Thu Jul 21 21:33:06 2022 daemon.err uhttpd[946]: stack traceback:
Thu Jul 21 21:33:06 2022 daemon.err uhttpd[946]: /www/cgi-bin/setup:656: 
in main chunk
Thu Jul 21 21:33:06 2022 daemon.err uhttpd[946]:        [C]: ?

aanon4, 2022-07-21:
Thanks. This should fix it.

closed 2022-07-27 by dman776