aredn/aredn #456: Ability to exchange routes with other networks.

netsound, 2022-07-25:
The ability to inject routes to networks not on ARDEN nodes but connected to such nodes.  These routes should be then be pushed onto the "ARDEN MESH" for other ARDEN nodes to be to be reached.   

Ability to support peering with other standards based networks via OSPF/BGP but hell one would even accept the ability to inject static routes for networks connected to an ARDEN node.

K6AH, 2022-07-25:
This will require some discussion.

aanon4, 2022-07-25:
You can already inject static routes into the AREDN network - not from the UI mind you - but OLSR supports it and I've "theoretically" done this on our local mesh in the past.

dman776, 2022-07-25:
what is the use case for this?   Can you expound about the idea with examples?

netsound, 2022-07-25:
So when I last tried to push routes while the local/directly attached node would add them it it's table they would not pass to other arden nodes meshed to that node.  It would only do RFC1918 space.  But 44 routes where note passed out.

@dman776   Use case is interconnection with other ham networks not running ARDEN.  Our primary Ham network runs native IP links with OSPF/BGP running  on 44/8 IP space.  But there are a few people who want to use ARDEN and we are in the spirit of allowing that to inter-connect.  We currently are able to setup and exchange routes via OSPF with an ARDEN node so that the IP space used on the mesh is pushed to one of our edge routers but trying to back the other way has proved the issue.  One way around that was to have that node annouce itself as a gateway but that doesn't work for large scale use or if the ARDEN network is connected to the large tunnel servers.

ae6xe, 2022-07-27:
@netsound   One way to do this today, with no modifications to AREDN firmware or 44net side of the fence is to setup a device on both the AREDN network and the 44net network.  then deploy a proxy to share services in both direction.   See this post:
https://www.arednmesh.org/comment/10632#comment-10632

Deploy a raspberry Pi to be on both networks.    Then deploy something like a Dante proxy server:
https://www.digitalocean.com/community/tutorials/how-to-set-up-dante-proxy-on-ubuntu-20-04

On the AREDN side of the fence the 44net services can be advertised from the Raspberry Pi.  On the 44net side the same native way of advertising services could be setup -- web page, etc. 

Further discussion would be needed and concerns addressed to integrate networks with no controls protecting resources between both sides of the fence.    This proxy approach would meet the use case of sharing services across AREDN and 44net networks in a controlled manner. 

Joe AE6XE

netsound, 2022-07-27:
@ae6xe   That then requires "Pi's" on both sides.. Requires users to then select the socks when they want to cross networks and requires the application to support the socks.   For an end user seems a bit more involved then just going to the IP of the service on ether network as a true ability to exchange routes would allow.  It would also mean less items be required for interconnection.    Not to mention speed hits for using the sock's proxy vs direct wire speeds.

As of now we can get having the AREDN node export via an OSPF connection mind you to a neighboring 44Net router is working.  And that AREDN node accepts the 44 route table back just fine.  It will however not share those routes to it's mesh neighbors.   A work around was to have node flag itself as allowing others to use it as a WAN.  But if there are other "WAN"'s on the mesh it's 50/50 the traffic replies will come back to that connected node.

Also doing the allow "WAN" route doesn't allow for multiple interconnections at different points in the network that have both an AREDN network and 44Network.

@dman776   Another real world example for ya.  Incident happening.  Users on site setup portable cameras on scene and because of the dynamic nature AREDN works great for this.  Tool in the Tool Box... The command post/trailer on scene has it's AREDN node to provide meshed IP networking on the incident. It also has a client radio connected to 44Net which brings it IP connectivity outside the scene to the larger WAN network.  IC asks for cameras to be setup at a few points around the incident so the task is assigned for the Hams to grab the AREDN portable camera kits and they setup around the scene.  At the command post since it's "on" the AREDN network it can hit each 10.x.x.x IP of the cameras and view them.  Well what about remote EOC that want's for view those cameras or control them directly. EOC is on the native 44Network and can talk directly to the AREDN's node WAN IP on site.  Sure one could go in setup the NAT rule in that node to a socks proxy like @dman776 suggested.  But now that fixed site viewer has to configure their browser to use the socks server and now all that users traffic is now routed to the socks on the incident scene even if it's not needed when they only needed to access a 10.x.x.x AREDN ip for one piece of traffic.

That isn't a theoretical but was a real use case incident. As AREDN is limited on what it can exchange with routes and also it's RFC1918 limits we were lucky that every place we were asked to put a camera could see a 44Net sector 5 or 6 miles away so the cameras were just on native 44Net IP's vs AREDN.   One did have a 20' mast on it though.   This allowed the cameras to be viewed at three different county EOC's and was even pushed to LTE first net devices for tablets on scene.   The onsite trailer to view the cameras had to have it's traffic go 5 miles off site and turn around but it worked.  

AREDN's ability for dynamic quick build meshed networks on an incident by appliance hams is it's strong point. Why it's a tool in the tool box for hams. It's easy for the non-network engineer types to setup and deploy useful tech on a scene or incident.  It is not suited well for large scale WAN networks.  Networks that need to work with other networks.  Networks that use IP's other then 10.x.x.x or 192.168.x.x. And that's fine it's why we have the other tools in the tool box to handle the big stuff. 

Interconnecting with other networks via a standards based protocol isn't new.  It's how the Cisco lovers talk to the Juniper lovers.   You wouldn't ask the Cisco guys to setup a SOCK proxy to allow the juniper networks to speak to it.  No you use BGP to exchange the routes you have on your network island with the routes the other network islands. It's what lets Cisco island and Juniper island residents talk to each other freely.  I am not saying support BGP.  BUT one could setup a port like you use DtD on Vlan5 on say VLAN6 that is OSPF able and exchange/tag the routes it learns via it towards the mesh table and vs versa.   Heck even if you started off with static non RFC1918 routes being able to be propagated that would be a start.  Baby steps.