aredn/aredn #464: [Feature Request] List of WAN wifi access points

aanon4, 2022-07-31:
List available access points when setting up wireless WAN.

k1ky, 2022-08-12:
While you're at it - how about a list of "connected" stations or an activity indicator in the LAN Access Point section? Or is there already an issue or should a new issue be created as a new feature/enhancement request?

aanon4, 2022-08-12:
> While you're at it - how about a list of "connected" stations or an activity indicator in the LAN Access Point section? Or is there already an issue or should a new issue be created as a new feature/enhancement request?

Could you make a separate issue for that? Thanks.

k1ky, 2022-10-21:
Done - new issue #529 feature request submitted.

cTurtle98, 2023-02-18:
this would be useful for me as well

km6lslf, 2023-03-19:
+1 this enhancement would be helpful