aredn/aredn #466: share wan on d2d (bridge between untagged on wan and vlan 1 on d2d)

cTurtle98, 2022-07-31:
on devices with a dedicated d2d port like the hap lite and gl-inet devices...

would it be possible to add a config checkbox to share wan with d2d.

internally i think this would be add a vlan 1 interface on the d2d port and bridge it with the wan interface. 
right now Im having to use a managed switch with the following config
untagged vlan 1 is wan uplink to my modem
untagged vlan 1 on wan interface on hap lite
untagged vlan 73 on lan on hap lite and tagged 73 on all other ports
tagged vlan 2 on all ports.

thats how I get an internet uplink to my other aredn devices at the site. 

if hap lite wan could be shared to vlan 1 on d2d then I could hook the wan port on it directly to the upstream internet and have a dumb poe switch on the d2d port connecting all my aredn radios at the site together

ae6xe, 2022-08-01:
Yes, this is possible and something I have an implementation for, but in a side branch with the openwrt 22.x.y upgrade branch.

cTurtle98, 2022-08-30:
> 

where can I download a sysupgrade for that?

ae6xe, 2022-08-30:
There's still work to make the upgrade branch functional.   Not enough yet to make a sysupgrade available and useful.

cTurtle98, 2022-08-30:
would it also be possible to add a toggle for d2d on wan.

use case: daisy chain hap ac lite with poe passthrough enabled

cTurtle98, 2022-12-27:
I have updated the issue name to be the name that would ve in the wan setting option in the advanced settings.

The point of this feature request is so that I can plug my site uplink directly into my hap or gl-inet device and still be able to use the administration tab to update my node

So I dont have to put my uplink into my switch then switch back to the wan on the hap or gl-inet

If this could be routed instead of a layer 2 bridge then it would remove the requirement for a non-aredn router just to do nat for all the nodes at a site where only one uplink ip is available

k1ky, 2025-10-06:
Isn't this a feature in the network options section? Allow MESH devices to use WAN / Remote WAN?

cTurtle98, 2026-01-04:
> Isn't this a feature in the network options section? Allow MESH devices to use WAN / Remote WAN?

that allows anyone on the mesh to use the devices wan, not just d2d devices.
I'm looking for a toggle to enable wan access for only d2d devices.