aredn/aredn #467: [Feature Request] Indicate whether WAN is coming from wifi or hardwired.

aanon4, 2022-07-31:
Tweak the language on the main page to indicate whether the WAN is wifi or handwired.

ae6xe, 2022-08-01:
bundled with this request, or possibly with another issue request...    The internet path may be down upstream, e.g. a home network router or the service provider may be the failure point.  The WAN gateway may still attempt to route traffic to the internet, but it never reaches.    A periodic check to e.g. 8.8.8.8 or other reliable internet service to determine if the internet is still reachable could be used to turn the gateway advertisement on/off and automatically reroute the traffic to  another gateway.

aanon4, 2022-08-02:
> bundled with this request, or possibly with another issue request... The internet path may be down upstream, e.g. a home network router or the service provider may be the failure point. The WAN [...]

Making this its own issue.

closed 2022-08-02 by dman776