aredn/aredn #468: Red link-established light illuminates on Ubiquiti nodes when no link is active

collierchun, 2022-08-01:
The center red status LED on several Ubiquiti illuminates even though no link is established or another AREDN node is within range.  The LED either lights steadily or cycles on and off, three seconds on, three seconds off.

This was first observed with firmware release 3.22.6.0  I also tried the 7/28/2022 nightly build and still saw the link status LED illuminate when no link was active and no other node within range.

I have seen this on several Ubiquiti Rocker M5s, a Ubiquiti PowerBeam 400, and several NanoStation M5s.   

i have attached support dumps from these three models, while running firmware 3.22.6.0 and nightly build 7/28/2022.

aanon4, 2022-08-02:
There are, unfortunately, many places in the code where the lights are manipulated - from the lowest level drivers all the way up. However, this particular behavior is correct by design and was introduced in 3.22.6.0. The 3 second on/3 second off flashing light is to indicate the node is ready, but not connected to the mesh. The light should go solid when a connection is established.

collierchun, 2022-08-02:
Oh cool, I've always waited until the boot light stopped flashing.  Good to know that the 3/3 flash is intentional.  I do believe I have seen the light on steady, though, with no link, and in fact, no signal on the SNR chart plots.  I will test again and get support dumps if I find that state. 

Thank you!

aanon4, 2022-08-02:
It can certainly be solid red for a minute or two after booting before the flashing starts.

collierchun, 2022-08-02:
Confirmed that the described behavior is occurring on my Rocket M5, PBE400 M5, and NanoStation M5.  The red link light initially lights continuously, but after a few minutes, the light starts to blink in the three second on/three seconds off pattern.  In all cases, there is no link established nor adjacent node present.  I'm agreeable to closing this as expected behavior/new feature.  

Thank you for reviewing this, 73s, and excellent work by the firmware team,
Collier
NM7B

closed 2022-08-02 by dman776