aredn/aredn #471: [Feature Request] Make the WAN more intelligent

aanon4, 2022-08-02:
From @ae6xe 

The internet path may be down upstream, e.g. a home network router or the service provider may be the failure point. The WAN gateway may still attempt to route traffic to the internet, but it never reaches. A periodic check to e.g. 8.8.8.8 or other reliable internet service to determine if the internet is still reachable could be used to turn the gateway advertisement on/off and automatically reroute the traffic to another gateway.

dman776, 2022-08-02:
how will this impact users who connect to a "larger" network that may not be "the internet"?   ie.   Mesh connected to an EOC network, but, isolated from the internet.

ae6xe, 2022-08-02:
Could have an adv option to define the 'internet' address to check (or local EOC address to check if reachable) -- default to something like 8.8.8.8 .   

Other detail would be in the implementation.  The criteria to take action?:

1) the local WAN interface should be "up" (wireless or hard cable) to do a  check, if not, should be no action to take.
2) For the greater mesh, if the gateway advertisement option is 'checked', then keep or remove the olsr advertisement based on confirming the path to the internet is working.
3) For devices on the node's LAN or programs on the node:  manipulate the node's default routing information to not direct any traffic to the WAN interface (might have to take WAN interface up/down?)

dman776, 2022-08-02:
i like the AdvConfig setting option to solve the "EOC" use case.

mathisono, 2022-08-10:
Stumped me once in the past when I had the local campus dns set on a node. Well that DNS is not reachable from the internet, so when my campus network port when down!!! DNS was unreachable from the redirected GW!