aredn/aredn #478: aredn.wan.vlanid=null In recent nightlies (1480-1598)

nc8q, 2022-08-10:
In some of my devices (GL-iNet, Mikrotik,...) there was no WAN IP even though
they should have gotten one from a DHCP server.
I noticed that aredn.wan.vlanid=NULL and
'Set to Default' did not place '1' in the text box.
Entering '1', then clicking ', then clicking reboot restored expected operation.
I was assuming that a 'Set to Default' would have put a '1' in the text box.
I think I always leave 'Save settings' enabled, so why did these devices lose the WAN '1' value?

73, Chuck

ab7pa, 2022-08-10:
Here is the information on the Adv Config VLAN setting:
https://arednmesh.readthedocs.io/en/latest/arednGettingStarted/advanced_config.html#advanced-configuration

> This feature only applies to node hardware which uses a VLAN tag for the WAN interface. It will not appear on hardware where the Ethernet ports are on a switch chip, since changing the default VLAN number is not supported on those devices at the present time. It will appear as a blank field on devices that have a dedicated WAN port and therefore do not need a VLAN tag for their WAN interface.
> 
> If you have node hardware that uses a VLAN tag for the WAN interface, then the default WAN VLAN identifier is 1. In some cases this default VLAN may be in use already or may be reserved by other equipment on your network. This field allows you to change the VLAN number being used on your node’s WAN interface.

nc8q, 2022-08-10:
Hi, Steve: Thanks.
My GL-iNet AR150 and AR750 are no longer obtaining a WAN IP address.
I will do more research.

closed 2022-08-10 by nc8q