aredn/aredn #479: Mystery Device

Orv, 2022-08-11:
I have a Ubiquiti  "Rocket M5 Titanium Ti GPS".   It's not listed in the Nightly Build README file and both the Rocket M and Rocket M XS software packages yield "[!!!! UNSUPPORTED DEVICE !!!!".   
I've attached the support data file.file:///home/obeach/Downloads/supportdata-NOCALL-202204170042.gz

aanon4, 2022-08-11:
This is the important snippet:
```
========== /usr/local/bin/get_hardwaretype ==========
rocket-m-xw
========== /usr/local/bin/get_boardid ==========
0xffff
========== /usr/local/bin/get_model ==========
Ubiquiti Rocket M XW
========== /usr/local/bin/get_hardware_mfg ==========
Ubiquiti
```
I'm concerned that the boardid comes back as 0xffff ... which is worrying non-specific. Does anyone know what the official Ubiquiti board id of this radio is?

Orv, 2022-08-11:
OK - this came to me used with a comment "Missing MAC?"   That lines up with the all ones board id, I think.  I'm closing this incident and tossing this device in the eWaste bin.

aanon4, 2022-08-11:
I went and looked at all the MACs. They are all 00:02:03:05:05:06 which seems a bit wrong.

Orv, 2022-08-12:
Closing this issue.

closed 2022-08-12 by Orv