aredn/aredn #486: Mesh Status forgets about links that are working

Orv, 2022-08-26:
I've seen this a couple of times before.  A high-level node says it's no longer seeing any RF neighbors, yet obvioiusly it's working as my node is able to connect to it.   Going to any other page and returning to mesh status restores the expected display.

[image]
[supportdata-K6PVR-VC-SimiEast-5G-202208251833.gz](https://github.com/aredn/aredn/files/9429359/supportdata-K6PVR-VC-SimiEast-5G-202208251833.gz)

aanon4, 2022-08-26:
So this is probably another case of OLSR updating a file and it being incomplete when this display is being generated .. which is why it then looks okay if you come back to it a moment later. We've seen this manifest in a couple of ways now so I should probably find a way to deal with it.

AJ6GZ, 2022-08-27:
Other behaviors seen: OLSR 0 Total / 0 Nodes on Node Status screen, and all-IP-only current neighbor entries on Mesh Status screen.  As noted above, both go back to normal upon refresh.

aanon4, 2022-09-09:
Did the recent nightly fix this for you?

Orv, 2022-09-09:
Looks OK, but doesn't occur routinely.  Will monitor.  Suggest closing issue for now.  Will open a new one if it reoccurs.

Orv, 2022-10-19:
This has been resolved in the nightly builds.

closed 2022-10-19 by Orv