aredn/aredn #489: Dropdown for Download Firmware on node doesn't show latest firmware.

AJ6GZ, 2022-08-27:
I don't use this feature, but it was brought to my attention... I don't know how long it's been broken.

On the Administration / Firmware Update section, after a refresh, the dropdown for Download Firmware does not contain recent firmware versions on several different node models (tested NSM, PowerBeam) and for some it does (SXT, possibly other MikroTiks).

[image]

aanon4, 2022-08-27:
If you reboot the node then hit refresh again does the problem persist?

AJ6GZ, 2022-08-27:
Same result after reboot.

dman776, 2022-08-27:
What is the download path on the advanced config page for firmware?   You may want to reset to default and re-save

AJ6GZ, 2022-08-27:
It was set to http://downloads.arednmesh.org/firmware/ubnt
Setting default changed to http://downloads.arednmesh.org/firmware/ 
Newer MikroTik already had this path.
Rebooted/Refresh and OK now.   Holdover from many upgrades ago I guess...
Will pass it on!

closed 2022-08-27 by AJ6GZ