aredn/aredn #490: BUG! Now run an hourly check on published service and “unpublish” any which aren’t really available.  Re-enabled services will be republished automatically.

mathisono, 2022-08-27:
All of the links I've set up to reach the Edgeswitches, look to be ghosted off the list. They must not respond to the check???

aanon4, 2022-08-27:
Can you add the support data for a node with this issue? Thanks.

mathisono, 2022-08-27:
Every hub that's got a edgeswitch linked, has had it dropped...  
[supportdata-KJ6DZB-PBR-10-202208271030.gz](https://github.com/aredn/aredn/files/9438411/supportdata-KJ6DZB-PBR-10-202208271030.gz)

aanon4, 2022-08-27:
So the problem appears to be HTTPS. You have a link which redirect to an https://.... link, and of course we don't support encrypted communications on the mesh, so this link isn't published.

aanon4, 2022-08-28:
Consensus is https links should be considered valid.
Fix: https://github.com/aredn/aredn/pull/491

aanon4, 2022-08-30:
@mathisono Can you verify this is fixed in the latest nightly?

aanon4, 2022-09-09:
Closing this as fixed in nightly.

closed 2022-09-09 by aanon4