aredn/aredn #492: Advertised services no longer publishing after update to 3.22.8.0

ai6bx, 2022-08-29:
I have been updating nodes and finding that advertised services, switches, phones, cameras, etc., are not showing when viewed under status. Going to DHCP/Forwarding, the links show up with a circle with a question mark in it. When save is clicked and processed, links come back.

aanon4, 2022-08-29:
Can you provide more information about the services which are not being advertised? Can you verify that the service can be reached (e.g. by entering the expected http://.... address). Please attach the support data for a node with these issues.

Changes in the new release will not publish services which are not actually available.

ai6bx, 2022-08-30:
Services include clickable links to access network switches, Astrisk PBX, phone portal, mapping software, etc. yes, they are accessible via IP and after I click the save button on the dhcp tab they become accessible and post visitability again. I will have to restart the nodes again to see if it recreates or if it only occurs upon upgrading.

Keith

aanon4, 2022-08-30:
So when you update the ports page, everything is immediately published regardless of whether the system thinks it exists or not. Only later does it check for validity. That's why you see them for a while, and then they disappear.

Do these links immediately resolve to their services, or is there some form of http redirection before that?

Also, can you please attached the support data? Thanks.

aanon4, 2022-08-30:
So, just looking at one of your nodes. Is this before or after services disappear?

[image]

aanon4, 2022-09-09:
Will close this shortly unless there's new info.

ai6bx, 2022-09-11:
Trying to upload screen shots here to with support file provided via email
[image]
[image]

aanon4, 2022-09-11:
Fantastic. Could you also provide the firmware version running on this node?

aanon4, 2022-09-11:
On the three entries indicating failures, when I test them I see the following:
1. Still investigating
2. I get a 404 when I request this page - which is considered non-valid and isn't published
3. I get 400 - Bad Request - Invalid hostname - which, again, is considered non-valid and isn't published.
4.

aanon4, 2022-09-11:
Update: On (1) the service uses a 307 redirect, which was missing in my code. I will add it and submit for the next nightly. Would be good to get feedback on (2) and (3).

aanon4, 2022-09-11:
https://github.com/aredn/aredn/pull/506

ab7pa, 2022-09-11:
Just to clarify, is this issue the result of specific non-AREDN hardware [returning HTTP codes](https://en.wikipedia.org/wiki/List_of_HTTP_status_codes) that we weren't expecting?  We expected standard 2XX "success" codes, while some cameras were returning 3XX "redirect" status codes.  This PR should allow the "redirect" codes to be interpreted as reachable on the network too.  Previous versions of AREDN firmware did not have the intelligence to check these services for validity, so this is definitely an improvement.

aanon4, 2022-09-11:
It's a very specific failure case. The validator code cannot validate https links because ... well .. ham radio doesnt like encryption .. so we don't and you can't specific https in the services list - only http. But what a lot of devices do is automatically redirect an incoming http request to an https request (self signed usually). We can't validate that (see earlier) so what we do is stop at that point and assume it's okay so long as the process to the https link is valid. That can include 3xx redirects, so when transitioning from http to https via a redirect, I check that the redirect is valid .. and there are more redirect codes these days that I was initially aware of. And I missed a couple.

ai6bx, 2022-09-11:
1709-e377086

ai6bx, 2022-09-11:
Looking into 2 and 3 now.

aanon4, 2022-10-18:
Please reopen if this is not fixed in the nightly.

closed 2022-10-18 by aanon4