aredn/aredn #493: Node names showing as IP addresses after update to 3.22.8.0

ai6bx, 2022-08-29:
After updating, all neighbor nodes show as IP addresses with link type as a question mark. A couple refreshes brings the node names back. Going back to the node shortly after, the nodes show as IP addresses again.

aanon4, 2022-08-29:
Closing as a known issue (already tracked).

closed 2022-08-29 by aanon4

Orv, 2022-09-09:
Doesn't occur routinely.  Will monitor.  Suggest closing ticket and opening new one if it reoccurs.

ai6bx, 2022-09-10:
This is happening quite frequently on my nodes.

aanon4, 2022-09-10:
@ai6bx Could you try the current nightly and see if it improves?

ai6bx, 2022-09-10:
Sure. I have attached the latest support file as well.

ai6bx, 2022-09-10:
Same with nightly.

aanon4, 2022-09-10:
Could you attach a screen shot and the nightly version you observe this on? Thanks.

ai6bx, 2022-09-11:
Here you go.

aanon4, 2022-09-11:
I'm afraid I dont see any screenshots.

ai6bx, 2022-09-11:
Try this one.

aanon4, 2022-09-11:
Possibly you need to do this on github rather than via email?

ai6bx, 2022-09-11:
Just uploaded under the issue request.