aredn/aredn #494: [Bug?] 3.22.6 & 3.22.8 : Configuration NOT saved!

VA2XJM, 2022-08-29:
After having updated our whole network to 3.22.6.0, I decided to go forward with the update of network switches. While doing those upgrade, the "main site" network switch changed the client MAC address used to administer the switch. So I went into the AREDN node management to change the DHCP reservation. This node is a hAP AC Lite.

Once change has been made, I clicked "Save" and always got "Configuration NOT saved". It didn't matter if I wanted to modify, remove or add a reservation.

I decided to test and go ahead toward 3.22.8.0 and the result is exactly the same. Finally I downgraded back to 3.22.1.0 and everything works as expected, I was able to change the MAC reservation and have the switch back under its hostname.

Any idea ?

aanon4, 2022-08-29:
Could you attach a screenshot showing the full page with error? Thanks.

VA2XJM, 2022-08-29:
@aanon4 I haven't take a screenshot since nothing out of the ordinary was present. I however have support data if you think you might find something in there that didn't catch my eye.

Since the node in question is the "master" of the network, I downgraded it asap to re-establish everything.

When the kids will be in bed tonight, I'll check our secondary tunnel node to see if the same happens or not.

aanon4, 2022-08-29:
What would be fantastic would be a screen shot of the node showing the error message, and then the associated support data at that time.  There are some cases where you cant save the configuration when it seems entirely reasonable, but they usually resolve around people switching the various wifi settings on and off. They are rare and fixing them hasn't reached the top of anyones annoyances.

This sounds different. There were bug fixes to the DHCP reservation because, if they were marked as non-propagating, they were essentially broken and only worked somewhat by chance (but mostly not at all). I'm wondering if this is somehow related - so a screen shot and then an immediate support data dump (so I can see the most recently log messages) would be great if possible.

VA2XJM, 2022-08-29:
I will try to reproduce in few hours.

Meanwhile for documentation purpose, The line that required the change, to the last digit of the mac (This is the actually working setup):
[image]

And joined is the support data package took right after the problem (tried change, delete and add) with all three displaying error.
[supportdata-VA2XJM-Gateway-202208290352.gz](https://github.com/aredn/aredn/files/9447995/supportdata-VA2XJM-Gateway-202208290352.gz)

aanon4, 2022-08-29:
As a workaround, did you try removing this entry then adding it back with the new mac address?

aanon4, 2022-08-29:
I can certainly manipulate a mac address on an DHCP entry on an hAP without a problem; so there's something more subtle going on. If you have chance to replicate this, please include a screen shot of the whole page - thanks.

VA2XJM, 2022-08-29:
First try (on 3.22.6), I tried to change the MAC. Got the error message. Rebooted.

Second try (on 3.22.6) to change the MAC, still got the error message. Reloaded the page, tried to remove the line, same error. Tried to add a new line (since the MAC is not the same), same error.

Updated to 3.22.8, then tried the same steps as the second try on .6.  Still no go.

Reverted to 3.22.1.0 since bed time was passed!

I will try to replicate as soon as possible and get screenshot + support dump.

VA2XJM, 2022-08-30:
Same on the "backup" tunnel hAP
[image]

[supportdata-VE2ATU-FN08LF-Gateway-202208292010.gz](https://github.com/aredn/aredn/files/9448544/supportdata-VE2ATU-FN08LF-Gateway-202208292010.gz)

VA2XJM, 2022-08-30:
BTW, I changed the last letter of the MAC to test.

aanon4, 2022-08-30:
So I can explain why your test doesn't work at least. You'll notice that both the IP addresses in your "DHCP Address Reservations" are already listed under the "Current DHCP Leases". If you try to change one of these entries to a new mac address, but  with an IP address which is already in use (which both are) it won't let you do that. Admittedly, some sort of error message would be useful here, so I will make sure that get's added before this issue is closed.

VA2XJM, 2022-08-30:
Well, in 3.22.1.0 and prior, we are able to change the MAC or the IP then save while devices are online. When the device renew it's lease, it receive a NAK and have to renegotiate a brand new lease with either the new assigned IP or a random one.

So, if I understand correctly, from 3.22.6, we need to wait for the lease to expire to be able to clear the reserved hostname and re-assign it to another device ? If so it means long waiting time with active "problems" into the dashboards when this happens AND lack of flexibility when changes needs to be done "now".

VA2XJM, 2022-10-13:
I took some time this morning to make various tests and I may have found a reason. I have the feeling that dnsmasqd is not working properly for unknown reason.

```
-> OLSR RESTART VIA WEBINTERFACE <-
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/all/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/tun56/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/tun55/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/tun53/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/tun50/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/eth1.2/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: Writing '1' (was 0) to /proc/sys/net/ipv4/conf/wlan1/send_redirects
Thu Oct 13 11:09:55 2022 daemon.info olsrd[3971]: olsr.org - 0.9.6.2-git_31f2f76cd5-hash_c00c03f009dce45c1dd8a9984d90e669 stopped
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '1' (was 1) to /proc/sys/net/ipv4/ip_forward
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/all/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/all/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/wlan1/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/wlan1/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Adding interface wlan1
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/eth1.2/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/eth1.2/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Adding interface eth1.2
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/tun50/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun50/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Adding interface tun50
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun52/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun52/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Adding interface tun52
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/tun53/send_redirects
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun53/rp_filter
Thu Oct 13 11:09:57 2022 daemon.info olsrd[6194]: Adding interface tun53
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/tun55/send_redirects
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun55/rp_filter
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Adding interface tun55
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/tun56/send_redirects
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun56/rp_filter
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: Adding interface tun56
Thu Oct 13 11:09:58 2022 daemon.info olsrd[6194]: olsr.org - 0.9.6.2-git_31f2f76cd5-hash_c00c03f009dce45c1dd8a9984d90e669 successfully started
Thu Oct 13 11:10:00 2022 cron.info crond[1165]: USER root pid 6256 cmd /usr/local/bin/fccid
Thu Oct 13 11:10:00 2022 cron.info crond[1165]: USER root pid 6257 cmd /usr/local/bin/rssi_monitor
Thu Oct 13 11:10:00 2022 cron.info crond[1165]: USER root pid 6258 cmd /usr/local/bin/snrlog
Thu Oct 13 11:10:00 2022 cron.info crond[1165]: USER root pid 6259 cmd /usr/local/bin/clean_zombie.sh
Thu Oct 13 11:10:02 2022 daemon.info dnsmasq[2946]: read /etc/hosts - 26 addresses
Thu Oct 13 11:10:02 2022 daemon.info dnsmasq[2946]: read /var/run/hosts_olsr - 41 addresses
Thu Oct 13 11:10:02 2022 daemon.info dnsmasq-dhcp[2946]: read /etc/ethers - 15 addresses

Thu Oct 13 11:10:10 2022 daemon.info dnsmasq[2946]: read /etc/hosts - 26 addresses
Thu Oct 13 11:10:10 2022 daemon.info dnsmasq[2946]: read /var/run/hosts_olsr - 60 addresses
Thu Oct 13 11:10:10 2022 daemon.info dnsmasq-dhcp[2946]: read /etc/ethers - 15 addresses

Thu Oct 13 11:10:15 2022 daemon.info dnsmasq[2946]: read /etc/hosts - 26 addresses
Thu Oct 13 11:10:15 2022 daemon.info dnsmasq[2946]: read /var/run/hosts_olsr - 95 addresses
Thu Oct 13 11:10:15 2022 daemon.info dnsmasq-dhcp[2946]: read /etc/ethers - 15 addresses

Thu Oct 13 11:10:21 2022 daemon.info dnsmasq-dhcp[2946]: DHCPREQUEST(br-lan) 10.208.61.39 00:0b:82:f3:1f:6d
Thu Oct 13 11:10:21 2022 daemon.info dnsmasq-dhcp[2946]: DHCPACK(br-lan) 10.208.61.39 00:0b:82:f3:1f:6d va2xjm-phone
Thu Oct 13 11:10:21 2022 daemon.info dnsmasq[2946]: read /etc/hosts - 26 addresses
Thu Oct 13 11:10:21 2022 daemon.info dnsmasq[2946]: read /var/run/hosts_olsr - 111 addresses
Thu Oct 13 11:10:21 2022 daemon.info dnsmasq-dhcp[2946]: read /etc/ethers - 15 addresses

[... cut ...]
-> Then nothing more from dnsmasq, just: rssi_monitor, snrlog, clean_zombie.
```
On all my other nodes every few seconds I see dnsmasq reading files. I pushed the firmware file once again just in case and it doesn't fix anything.

This node is, for now, the network central hub so it is hard to make tests without breaking the network. I'm trying to see what can have caused it with a lab node, replicating original install with 3.19.something and upgrading... maybe something broke. To be continued...

VA2XJM, 2022-11-27:
Ok, I found the problem with that bug report! Tonight I decided to "remake" the whole node due to other problems I was investigating.

Look at the last screenshot from Aug 29. The last port redirection line contains a range (12000-14500). When removing "-14500" from the port range I need open, it saves perfectly. In the previous versions it was opening the whole range specified and in the LAN port only the first port of the range has to be specified. 

See docs:
```
Port forwarding will redirect inbound connections from the Wifi, WAN, or both interfaces and forward them to an IP address and port on the LAN. The destination port number need not be the same. If you have hosts on the LAN that provide services you want to make available to the mesh, a Port Forwarding rule will be required to make that connection possible.

If you want to forward a range of ports, the Outside Port field will accept a hyphen-separated range in the form "xxxx-xxxx". When doing this, set the LAN Port to the low value of the port range. When forwarding a port range the Outside and LAN ports must match.
```

VA2XJM, 2022-12-17:
@aanon4 I'm not a LUA dev but I've spent an hour this morning tracking that bug... I think I found it, and I pushed a PR to fix it, but first I'd like to have your input on that...

So I compared the `/files/usr/lib/lua/aredn/utils.lua` V 3.22.1 and V 3.22.6. This is were the code switched to LUA. I then switch to latest and isolated the `validate_port_range` function and played with it in [jdoodle](https://www.jdoodle.com/execute-lua-online/).

Here is the actual code in the AREDN repo:
```
function validate_port_range(range)
	if not range then
		return false
	end
	local port1, port2 = range:match("^%s*(%d+)%s*-%s*(%d+)%s*$")
	if not port2 then
		return false
	end
	if not validate_port(port1) or not validate_port(port2) then
		return false
	end
	if tonumber(port2) > tonumber(port1) then
		return false
	end
	return true
end
```

To make it return a valid range (used "1000-1024" to test), I had to make 2 changes.

1. `if tonumber(port2) > tonumber(port1) then`  BECOMES `if tonumber(port1) > tonumber(port2) then`
2. `local port1, port2 = range:match("^%s*(%d+)%s*-%s*(%d+)%s*$")` BECOMES `local port1, port2 = range:match("^(%d+)%-(%d+)$")`

To test it out, I've wrote the changes directly into a node through Telnet and tested it. The GUI return a success message and the range is well saved:
[image]

Not having the whole code in head, I just want to ensure there is not negative impact somewhere else, but I believe this might well be the fix.

closed 2022-12-22 by dman776