aredn/aredn #497: Channel 6 disallowed for 2GHz LAN AP mode

ab7pa, 2022-08-30:
As reported in the AREDN forum, several people are having issues with the **LAN Access Point** failing if they use the 2GHz radio on channel 6.  The issue appears across hardware vendors (GL.iNet, Mikrotik, Ubiquiti).

_hostapd_ correctly starts and runs on all other channels, but the attached file shows the logread messages for _hostapd_ when using channel 6.  There are some messages in the OpenWRT forum about disallowed channels, but nothing specific to our issue of using all channels except channel 6.
[hostapd-logread.txt](https://github.com/aredn/aredn/files/9453666/hostapd-logread.txt)

ae6xe, 2022-08-30:
in the log file:
"IEEE 802.11 Configured channel (6) not found from the channel list of current mode (1) IEEE 802.11g"
What is unusual is that we want the AP to be using 802.11n mode, then revert if 1 of the multiple connecting clients is only 802.11g.   Nothing in the AREDN patches appears to impact the use of ch 6 -- it's in the middle of all the regulatory and country code settings.      Wild guess, but I'd test by removing 'htmode' and/or 'hwmode' options, to fallback to a default, and see what happens.  Remove from /etc/config/wireless and reboot (this is the section on an hap ac lite where 2GHz is radio1):

config wifi-device 'radio1'
        option type 'mac80211'
        option disabled '0'
        option channel '6'
        option hwmode '11g'
        **option htmode 'HT20'**
        option path 'platform/qca953x_wmac' 

Also what does this output look like on a node that fails?   "iw reg get"

ab7pa, 2022-08-31:
Good thoughts.  I removed both the _hwmode_ and _htmode_ options from /etc/config/wireless and rebooted.  Same behavior with channel 6 not allowed and hostapd disabled.

iw reg get for 2.4ghz = (2400 - 2472 @ 40), (N/A, 30), (N/A)

When I look at the auto-generated hostapd-phy0.conf file it seems to have defaulted to hw_mode=g even without specifying it in /etc/config/wireless.

ae6xe, 2022-09-05:
This is related to: https://github.com/aredn/aredn/issues/39.    When we added 2 more channels (-3 and -4), the symptoms shifted 2 channels from 8 down to 6 (note the correlation of adding 2 channels and shifting symptoms down 2 channels).   Good chance I found the code that needs updated that explains this.

dman776, 2022-12-22:
@ae6xe didn't this get fixed already?

ab7pa, 2022-12-22:
I just tested this again.  All 2.4ghz LAN AP channels are usable **_EXCEPT_** for channel 6.  Channel 6 is not available when using 2.4ghz LAN AP.

ae6xe, 2022-12-27:
Ch 6 failure still exists that I know.   Note, before adding ch -3 and -4, Ch 8 was failing.   Might still be an open issue on ch 8?

Adding 2 channels,  and then symptoms shifting 2 channels is interesting, maybe a clue to root cause.

dman776, 2023-04-19:
@ae6xe @aanon4 did we fix this?

aanon4, 2023-04-19:
Not knowingly unless it got fixed by the latest openwrt version.

ab7pa, 2023-04-19:
This is very easy to test.  I'll test it again tomorrow after our family visitors are gone.

ab7pa, 2023-04-19:
I just tested this again (20230419 0700mst). All 2.4ghz LAN AP channels are usable EXCEPT for channel 6. Channel 6 is not available when using 2.4ghz LAN AP.

ab7pa, 2023-04-19:
Channel 6 appears to be working when the 2.4ghz radio is configured as a LAN AP on my AR750. I now see it in the list of wifi APs on a couple of my 2.4ghz wifi devices in the shack.  THX