aredn/aredn #530: N-1835 Doesn't show Service Links for older nodes running Firmware 3.16.1.0

k1ky, 2022-10-22:
Bug: Service listings on older nodes running 3.16.1.1 don't show in the node list on nodes running Nightly 1835 and the previous nightly that introduced the new sortable GUI feature.

aanon4, 2022-10-22:
Could you post some screen shots? Specifically, a screen shot on the nightly showing the missing services, one on 3.22.8.0 showing the services, and one of the services on the 3.16.1.1 node.

aanon4, 2022-10-22:
And if possible, support tool data for each of the nodes. I realize that might be difficult - but at least the nightly node and the 3.22.8.0 node.

k1ky, 2022-10-22:
This is the N-1835 node screen shot.  The 3.16.1.1 node is KM4ACK NSM5XW node below that doesn't show the services:
[image]

aanon4, 2022-10-22:
Does it show the correct services when you're not filtering by its name, or only when you are?

k1ky, 2022-10-22:
Actually it's the KM4ACK-ARHP-200 services that don't show:
Here is a view from a node running N-1757
[image]

aanon4, 2022-10-22:
In your example, KM4ACK-ARHP-200 doesnt have any service directly hosted by the node. It does have services attached to LAN connected machines, but if you're using the node filtering option, these are not selected as they're not nodes.

k1ky, 2022-10-22:
Support data files

> 

You are correct - If I don't use the "filter", the services display properly:
[image]

k1ky, 2022-10-22:
Support data with renamed extensions - one was a tgz and the other was gz.
[supportdata-KM4ACK-ARHP-200-LASCASSAS-202210212101.zip](https://github.com/aredn/aredn/files/9843341/supportdata-KM4ACK-ARHP-200-LASCASSAS-202210212101.zip)
[supportdata-K1KY-HAPA-450-P2-202210212100.zip](https://github.com/aredn/aredn/files/9843342/supportdata-K1KY-HAPA-450-P2-202210212100.zip)

k1ky, 2022-10-22:
> 

Apparently it's only an issue when using the filter.  Maybe this is acceptable behavior and I just don't understand.  I figured it would still sow everything for the selected node.

ab7pa, 2022-10-22:
The services you want to view seem to be provided by the LAN Hosts on that node.  If that is the case, then you could search in the **LAN Hostname** column for KM4ACK.  I can see, however, that even that search will not pick up everything because KM4ACK has not prefixed his LAN hostnames with his callsign.  According to the "best practice" suggestion in the online docs, even the LAN hostnames should be prefixed with the owner's callsign in order to give it a unique name across the entire mesh.  Hope this helps.
From the [Port Forwarding, DHCP, and Services](https://arednmesh.readthedocs.io/en/latest/arednGettingStarted/advanced_config.html#port-forwarding-dhcp-services-and-dns-aliases) section:
[image]

k1ky, 2022-10-25:
Yes, that is correct.  All services and node information show when "not" sorting.  One would expect that it should show the same information when sorting?  I believe that is the issue that I was attempting to highlight.

ab7pa, 2022-11-03:
@k1ky Tom, based on the screenshots you provided it appears that the **Mesh Status** column filtering feature is working exactly as intended.  The issue has nothing to do with which firmware version the remote nodes are running.  You asked the page to show you any node names containing "km4ack" and that's exactly what is shown in your screenshot.  
[image]

If you want to see the LAN services computers which have "km4ack" then you should search in the _LAN Hostname_ column.  Here is an example from my local mesh of searching in the _LAN Hostname_ column for service computers.  This works perfectly because this node owner properly prefixed his callsign to his service computer names.  If KM4ACK had prefixed his callsign to his service computers then you would have found all of them just like in this example.
[image]

PR #546 (2022-11-10): fix mesh search rows
Address the enhancement request in issue 530.
Fixes #530
closed 2022-11-10 by dman776