aredn/aredn #539: Integrate a new Target(ramips) and Subtarget(mt7521)

RRERF, 2022-10-28:
Hi there,

I'm trying to load the aredn - openWrt functionality to a MikroTik RB750Gr3. We want to use it as VPN-Gateway to our AREDN-Network over Internet and over HamNet.

Can anybody help me please?

ae6xe, 2022-10-28:
The aredn firmware, for now, expects the device to have a wireless card.  Some work is necessary to update all the code-locations where calls made to get the wireless info -- status, setup, and etc.   We can take this as an enhancement request.

ae6xe, 2022-10-28:
As an alternative, and usable now,  the HAP AC Lite model 952Ui-5ac2nD is same footprint and similar device commonly used by AREDN community.   There is a setup option to turn off the wireless use and effectively the device would be the same as this hEX device.    It is lower memory and Ethernet speeds, however.

RRERF, 2022-10-29:
Thank you for answering.
Is there a Configuration-Script available for OpenWRT, so i could cut all Settings, what are not used...?

k6ccc, 2023-02-20:
The Mikrotik RG750gr3 is a fine router - I used one for years at my house as my primary router (it's now a backup in case the RB4011 fails).  However it would seem a poor choice to try to force it into AREDN service.
As Joe mentioned, the hAP ac lite is extremely commonly used for home and portable service for AREDN and has a very solid track record.  And if you want faster Ethernet connections and or more memory / horsepower, there is the hAP ac2 and hAP ac3 - both of which are now supported in nightly builds for AREDN.  I replaced a hAP ac lite with a hAP ac3 last weekend as my primary AREDN node at home.  Working great.