aredn/aredn #540: TP-Link cpe210 v2 Nightly build DNS refused

KO4BJV, 2022-10-28:
Hello. Sorry if this is already known . 
Firmware [aredn-1853-91cf71f-ath79-generic-tplink_cpe210-v2-squashfs-sysupgrade.bin](http: 22-10-27

When clicking on a link to a neighbors node  the status page of that node will not load. It can be reached by its ip address. From Wireshark DNS query reports refused A ( name of node ). Rolled back to stable image DNS working okay now.

Kyle

Orv, 2022-10-31:
Have you verified that you only have one network connection to your computer?  That is, if it's a laptop and you're wired to the TP-Link, is the wireless turned off on your laptop?  That's a frequent source of DNS issues.

Orv W6BI

KO4BJV, 2022-10-31:
I did. I have a wifi router set up as access point for the node that I ran the nightly build on. The router is directly connected to the node and its set default to hand out 5 ip's. I did get ip address and DNS server was the nodes gateway. I did clear a few weeks of cache from my laptop browser. It seemed like it couldn't resolve the neighbor nodes by there FQDN if that's the correct term. I tried ping from cli but failed by name.  With the stable firmware I can ping and use nslookup  by using ( ko4bjv-node-2.local.mesh ). I guess this is an accurate way to test. It may just be some mistake on my end. 

Aredn is an amazing network. I'm in the flat land here on the east coast so the trees are an issue but it still works reliably for my needs. Thank you to all the developers and contributors. 

Kyle

Orv, 2022-11-05:
I put the nightly build on my CPE210-v1 yesterday and found no issues.   Dunno what's going on...

KO4BJV, 2022-11-09:
Okay Thanks for checking. I must have been doing something wrong on my end,

closed 2022-11-09 by KO4BJV