aredn/aredn #555: [Bug] Tunnel settings are not refreshed when renaming a node

VA2XJM, 2022-11-27:
While doing network maintenance jobs, I needed to rename a distant tunnel client node to reflect a site callsign change. The node name has been also changed on the tunnel server side. Both from the GUI.

After the client node has been rebooted, the tunnel will not go up. On both side, we can see the connection is rejected.

I logged in into the client node, edited /etc/config/vtun and /etc/config.mesh/vtun to change the callsign portion of the "name". Rebooted the client and now the tunnel will go up.

---

Changes needed: When the node name is changed, the tunnel "name" values for tunnel clients must also be changed. Those values could be computed at every reboot to make it simple.

k1ky, 2022-12-04:
Ditto - I have experienced this issue for over 6 years.  I just reported it on the kg6wxc MeshLink as well.  Changing the name back on the tunnel server side seems to restore the connection.  Thanks for digging into this.  I look forward to a fix as I need to rename more tunnel client nodes.

k1ky, 2022-12-04:
It appears that something that I have done - disabling/reenabling server side connection, rebooting server and client nodes - the connection now works under the new name.  I checked the /etc/config files listed above on the client node and the new name appears to be correct in those files.  I had not edited these files, so "something" magically corrected the situation after many hours.

ab7pa, 2022-12-04:
@k1ky Tom, from what I see in the code, every time you click **Save Changes** on the _Tunnel Client_ page all of the config settings are written to those config files and both oldrd & vtund are restarted.  There is no field in the UI to enter the nodename, but under the covers the nodename is pulled from the current name on the node.  So if you change the nodename of a tunnel client then you should also go to the _Tunnel Client_ page and click **Save Changes** (even though it looks like you're not making any changes).  That might provide a workaround for you.  Let me know if that works.  In the meantime we'll see if there is a way for that "magic" to happen directly from the _Setup_ page when you change the nodename on a tunnel client.  On the tunnel server side there is a field for the Client nodename, so that should be changed and then click **Save Changes** on the _Tunnel Server_ page.

VA2XJM, 2022-12-05:
@ab7pa Thanks for that finding. :)