aredn/aredn #558: Mikrotik HAP ACL doesn't render entire first screen below the buttons N-1948 to N-1955

k1ky, 2022-12-06:
Some (not all) Mikrotik HAP ACL will not render the entire first GUI screen below the Buttons.  All other screens appear to be fine.  So far this has occurred on 3 nodes out of probably more than 15 that have been upgraded.

k1ky, 2022-12-06:
[image]

k1ky, 2022-12-06:
[supportdata-W4HPL-HAP-150-WETZEL-202212051254.gz](https://github.com/aredn/aredn/files/10163503/supportdata-W4HPL-HAP-150-WETZEL-202212051254.gz)

AJ6GZ, 2022-12-06:
I noticed an unclosed </nobr  tag on line 287 of /www/cgi-bin/status but fixing it on the affected node didn't help.  Also that typo looks to have been there at least since 3.22.8.

Output html just seems to stop at this point on his node:

[image]

k1ky, 2022-12-06:
It appears that having Mesh RF disabled is the key.  Enabling Mesh RF - even with no connections available seems to fix the issue.

k1ky, 2022-12-06:
MESH RF=Enabled 
[image]

ab7pa, 2022-12-06:
Thank you @k1ky Tom and @AJ6GZ Ian for finding this.  I made a fix for this issue and the new **/www/cgi-bin/status** file is available here if you want to download it and test it:
https://raw.githubusercontent.com/aredn/aredn/f7420ee0629236a0198cf013f11293bb37e56f8b/files/www/cgi-bin/status
I'd appreciate your feedback.  Many thanks!

dman776, 2022-12-06:
> You betcha Tim - I may wait for the nightly fix. Maybe Ian is up for it!?

(That was Steve that fixed it, not Tim….).    😉

kc2cbd, 2022-12-06:
That fixed the issue Steve.
Thanks

PR #559 (2022-12-06): fix status page not rendering
Reproduced Issue #558 and provided a fix for that issue as well as the unclosed _nobr_ tag. Fixes #558
closed 2022-12-06 by dman776