aredn/aredn #582: MikroTik RouterBOARD LHG 2nD-XL refuses to take Nightly 2024

k6ccc, 2022-12-24:
Updating my assorted collection of nodes to Nightly 2024.  So far, all have gone well except a LHG 2nD-XL.  Staring from Nightly 2002 and using the aredn-2024-fa898cc-mikrotik-lhg-2nd-sysupgrade.bun file.  Node appears to update normally, but comes back to 2002 and uptime does not show that it rebooted.  Repeated many times and have manually rebooted and power cycled with no change.  My speculation is that because the update page shows that it is a hardware type of rb-lhg-5hpnd-xl (always has), that the update is showing the incorrect firmware type and rejects the update.  Support file attached.
[image]
[supportdata-K6CCC-RB-LHG-2nD-XL-202212241304.gz](https://github.com/aredn/aredn/files/10299143/supportdata-K6CCC-RB-LHG-2nD-XL-202212241304.gz)
[image]

aanon4, 2022-12-24:
You wouldnt have a link to the specific nightly this is currently running would you?

For some upgrades which are going from a generic install to a specific install, the upgrade fails to start because it thinks the image is incompatible. We can fix this, but it would be easier if I had a copy of exactly what it's running.

k6ccc, 2022-12-24:
It was Nightly 2002 to start with.  I might still have a copy of that floating around...

k6ccc, 2022-12-24:
Should have been:  aredn-2002-29d1251-ar71xx-mikrotik-rb-nor-flash-16M-squashfs-sysupgrade.bin
That was downloaded on my work laptop which I can not access remotely.

The plot thickens.  My LHG 5ndHP-XL also did not update.  I missed the fact that that node also did not update.  Guess I was going to fast.  It also was Nightly 2002 - using the same file.  Support data for that one attached.
[supportdata-K6CCC-RB-LHG-5HPnD-XL-202212241517.gz](https://github.com/aredn/aredn/files/10299252/supportdata-K6CCC-RB-LHG-5HPnD-XL-202212241517.gz)

aanon4, 2022-12-24:
Each build contains information which says what it's compatible with, so it wont install incompatible firmware (which is nice). I will need to add this compatibility information to the relevant new builds so they know they're compatible with these old generic builds .. I just needed the generic so I can extract that info (rather than guessing what it probably is).

k6ccc, 2022-12-24:
Man, I really was not paying attention.  The LHG 5nd also did not update.  It also had been 2002.
[supportdata-K6CCC-RB-LHG-5nD-202212241533.gz](https://github.com/aredn/aredn/files/10299261/supportdata-K6CCC-RB-LHG-5nD-202212241533.gz)

AJ6GZ, 2022-12-29:
Report: LHG-5HPnD-XL as of build 2045 updated OK.

ab7pa, 2023-01-04:
My LHG-5HPnD will take the NB firmware, but only if I use the "Load Local Image" method for the firmware install.  The typical "Upload Firmware" process displays the upgrade message, but reboots to the previous firmware image without ssh enabled.  A manual reboot returns the node to working state on the previous firmware image with ssh.  Then I can scp the image to the node and use the "Load Local Image" process to accomplish the upgrade without issue.

aanon4, 2023-01-04:
@ab7pa Whatever's going on with your node is odd. I can upgrade my own LHG-5HPnD without a problem.
Can you describe your process when it fails? Particularly, how are you connected to the node you're upgrading? Over the mesh? Over wifi. On the LAN? The actual upgrading command is the same for both your failing case as well as the one which succeeds, so whatever is failing is before that. Ping me online so we can disable the part of the upgrade process which disables logging so we can get some more info.

k6ccc, 2023-01-04:
This was resolved a week or so ago with the introduction of separate files for 2GHz vs 5GHz.  All three of my LHG nodes have been updated several times since then successfully.

ab7pa, 2023-01-04:
Thanks, @aanon4 Tim -- will work with you offline.
If this issue is resolved we can close it.  THX

closed 2023-01-12 by aanon4