aredn/aredn #586: Mikrotik QRT5 RB911G-5HPnD-QRT

kc2cbd, 2022-12-25:
Are we going to get firmware for the Mikrotik QRT devices?

aanon4, 2022-12-25:
This hardware is not supported by the current version of OpenWRT, so someone would need to donate hardware so we can do the port.

kc2cbd, 2022-12-25:
This device used the old Large firmware.

AJ6GZ, 2022-12-25:
Are we talking the N or AC version? The N was previously supported.

aanon4, 2022-12-25:
The N version was supported on the old ar71xx target which OpenWRT removed. Everything needs to be ported to the ath79 target and this has not been done for this device.

That said, the 911 devices are very similar to the 912 devices, and I did the 912 devices already. So .. if someone wants to donate hardware it can probably be done (see my original comment).

kc2cbd, 2022-12-26:
I don't know which one it is. Can't tell by the specs on Mikrotik's site.

AJ6GZ, 2022-12-26:
RB911G-5HPnD-QRT   "QRT 5"   N version. Supported on 3.22.12.0
RB911G-5HPacD-QRT  "QRT 5ac"  AC version.

If you already have one working, it's the N version.

kc2cbd, 2023-01-30:
Firmware CANNOT be updated
firmware file is not valid

`for i in /tmp/sysinfo/* ; do echo $i ; cat $i; done`

kc2cbd, 2023-01-30:
Firmware CANNOT be updated
firmware file is not valid

`for i in /tmp/sysinfo/* ; do echo $i ; cat $i; done`

`/tmp/sysinfo/board_name`
`rb-911g-5hpnd`
`/tmp/sysinfo/boardid`
`Mikrotik RouterBOARD 911G-5HPnD`
`/tmp/sysinfo/hardware_mfg`
`Mikrotik`
`/tmp/sysinfo/model`
`Mikrotik RouterBOARD 911G-5HPnD`

AJ6GZ, 2023-01-30:
Confirmed same error here.  Same strings as above. (This is the N version).

aanon4, 2023-01-31:
As noted in the supported device file, you cannot update but must reinstall. This is a limitation on all nand based Mikrotik devices transitioning from the older version of openwrt.

AJ6GZ, 2023-01-31:
Reinstalled without issue. So far so good on live mesh.

kc2cbd, 2023-01-31:
Same as AJ6GZ...Reinstalled as a new device without issue and installed iperfspeed as well. Seems to be working correctly.

aanon4, 2023-01-31:
Thanks - looks like I messed up where I put the re-install in the SUPPORTED DEVICES file - will fix it.

closed 2023-02-08 by aanon4