aredn/aredn #588: Tagged Vlan ID's for ports on hap-ac2, 5 port, and 3 port devices

mathisono, 2022-12-25:
The default tagged port config for the hap-ac2 is exactly the same as the original hAP.  Only no POE passthru!  

Im requesting that port 1 (POE in) be treated as a trunk port like a single port . Im hopping that by default (after first boot) vlan2 & Vlan1 would be are set as tagged traffic on the on the first port. I uses EdgeRouters/switches, POE is distributed for nodes from a EdgeSwitch. Ive dedicate 2 ports to a GL-AR750 to integrate it with other single nodes for dtd and another port to supply WAN.  Now with the hap-ac2 support Im planning to replace a NanoStation with a hap-ac2. Powering the hap-ac2 just like I do the NSM5. The rest of the ports will be used for the mesh Lan.  

Preference isn't to hack the node every time I upgrade with the changes. Im requesting a method in "Advanced Configuration" to toggle between port layouts. Im suggesting a toggle between port layouts. Or another option would be to allow user to select Tagged DTD vlan, tagged WAN vlan, Untagged Lan.  Check mark enable box for specific ports.

[image]
[image]
[image]
[image]
[image]

aanon4, 2022-12-25:
I dont disagree with the idea that we need more flexibility here - at the moment I have 2x hap-ac2 with unique configurations, but in my case I want extra DtD port and less interested in more LANs. There is a mechanism (file edits) to customize this today on the hap-ac2 and these changes persist between upgrades, but I take the point that something more UI might be good. The trouble is what is the other configuration? For example, I'm plugging an hap-ac2 into a edgepower right now, and using it's PoE output to power the HAP and making that port just a LAN so I can monitor it.

Maybe the alternate configuration should have all the ports the same rather than a different set of special ports?

mathisono, 2022-12-25:
Consider this toggle between a preset with Tagged dtd vlan and untagged local lan on interfaces. Provides a fall back scenario to access the node.  Ubiquity provides the configuration option to exclude undefined vlans. 

[image]
[image]

AJ6GZ, 2022-12-25:
Perhaps we could come up with a few preset "modes" or templates depending on common end user scenarios. Here are a few I came up with.

Traditional/PoE
WAN     LAN      LAN      LAN      WAN/DtD

Resource Host
WAN/DtD/LAN(u)    LAN      LAN      LAN      LAN

Router Only
WAN/DtD/LAN(u)  WAN/DtD  WAN/DtD  WAN/DtD  WAN/DtD

(u)=untagged

The Services one is what I would use, providing a stable place to park resources without having IP's change if a tower node needs replacement, etc. This would be for those with a VLAN capable switch of course with the benefit of only using 1 port for uplink.  Note that LAN is retained on the trunk port too for expansion.  I know several people who want this one.

Ideally we'd like to have each port fully configurable rather than an ever-growing list of templates or mode buttons...  But we'll have to balance the extra complexity and admin lockouts vs. it-just-works if we make it "too" easy to find all 60 permutations.  Also note that just having all ports with all 3 VLANs will introduce DHCP overlaps.

Lots to think about!

mathisono, 2022-12-26:
Im not suggesting all 3 VLANs be introduce on all of the ports!!! That's I feel that for a novice user without any background on configuring vlans, would be exposed a "select mode of operation" to configure based the users/site specific configuration.  Template based configuration doesn't lock an advanced user into ONLY this or that preset. For novice users it an learning opportunity to see the veristically of a node configuration. 

@aanon4 could you expand on the topic of "There is a mechanism (file edits) to customize this today on the hap-ac2"  

Q: Would you share this (file edits) method...

Consider 1: prewriting Vlan port templates with the option and wiki to support a user writing their own, based off the, known to work templates. 

Consider 2: Write a GUI to set tags on specific ports. As @aanon4 said "something more UI"

aanon4, 2022-12-26:
So there are some, currently undocumented, configuration files to persist configurations for swconfig based devices as well as the new hap ac2 and ac3. Why undocumented? Well they're very new (especially the ac2 and ac3), but we've yet to decided how to best document stuff like this. Most users will never need this, and its very much "aim gun at own foot" territory, so there's a balance here between accessibility and the support burden it will generate. That said, I dont like undocumented features, so we'll need to solve this problem somehow.

> Im not suggesting all 3 VLANs be introduce on all of the ports!!!

Actually what I thought you might be suggested is that we could have a toggle for each port to be either "the default" (as it is configured now) or 2 VLANs + untagged (like a single port device would present). I dont know if this is confusing or not nor what the UI would need to look like.

mathisono, 2022-12-29:
https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial

GUI work fine in the main line build...

mathisono, 2022-12-30:
I concur, a toggle for dtd for each port. would be simple... Its counter to the way the config handles this tho... Ports get assigned to vlans not vlans to ports. The gui should at least expose the novice user to the concept of ports being assigned to vlans and traffic being T or U.  The one situation that could be tricky, would be if some one wants use the wan port to pass MeshWan as a T rather than U on a port. Its a possible way to configure the device.  Only the (MeshWan) should have the option to assign different VLAN id (as it is now). Would add rows too the "advanced config page" with a new section called "vlan settings" DtDlink / MeshWan / Lan, each with a check box to assign the devices ports wan,lan1,lan2,lan3,lan4. Only presents the option to set a T/U state for the MeshWan's vlan. Should the T/U state be port specific for MeshWan's vlan id ? Or be blanket. That wouldn't accommodate the situation above.

Thoughts on the method?

BTW 

Re: hap ac2 and hap ac3,
If you look in /etc/aredn_include youll see some files ending .network.config. Change the .config files to reflect the new config. Copy/write these files so you have a version ending with .user and .config - e.g. dtdlink.network.user & dtdlink.network.config in this locations. Each starting with the particular network name (so wan, lan or dtdlink).  
This is changing the operation of the physical port on the device!

To persist these changes across upgrades you need to copy these files to their .user versions otherwise they will not persist across updates. Copy/write these files so you have a version ending .user and .config - e.g. dtdlink.network.user & dtdlink.network.config!

When you've made the changes you need, run /usr/local/bin/node-setup -a mesh to regenerate the actually network files used to configure the system on boot and then reboot the node. Obviously if you get this wrong and lock yourself out of the node.

cd /etc/aredn_include
vi /etc/aredn_include/dtdlink.network.config

Add the needed lines, then copy/write the 2 files. 

/usr/local/bin/node-setup -a mesh
reboot

Here is what my config looks like:
---
# cat wan.network.config
        list ports 'wan'
----
# cat lan.network.config
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        
----
# cat dtdlink.network.config
        list ports 'lan4.2'
        list ports 'wan.2'
----

cTurtle98, 2022-12-31:
what about a checkbox table?

| VLAN | Port 1 | Port 2 | Port 3 | Port 4 | Port 5 |
| --- | --- | --- | --- | --- | --- |
| WAN |  [X]  |  [ ]  |  [ ]  |  [ ]  |  [ ]  |  
| LAN | [ ] | [x] | [x] | [x] | [ ] |  
| D2D | [ ] | [ ] | [ ] | [ ] | [x] |  

lan enabled on port 1 = wan tagged
lan disabled on port 1 = wan untaged

this would also contribute towards solving #466 
but it wouldn't solve only 1 ip being available on wan and needing to route the wan connection like is done for allow mesh to use wan

k1ky, 2022-12-31:
My most common configurations are: HAP:  DTD5, DTD4,5, DTD3,4,5  and AirRouter: DTD4, DTD3,4, DTD 2,3,4 and DTD1,2,3,4
I usually leave at least on Lan port available to connect a regular device (non-DTD). If I need more, I'll go with an external switch for more DTD Ports, and Lan Ports. The swconfig.user file has been a great timesaver for device upgrades.  I usually denote in the device on-screen description file what the state of the ports are: DTD4,5.u (or whatever).  The .u indicates the existence of the config.user file so I know that the upgrade covers the custom Port configuration for that node.  I haven't messed with the WAN port settings so far up until now.

mathisono, 2023-01-01:
The checkbox table? would be great! Consider a 3 way slider |off/T/U| for the WAN.  

List ports 'wan.2' for DtD helps me integrate with my core switches!

aanon4, 2023-02-05:
Some sort of checkbox table like this?

.  | wan | lan1 | lan2 | lan3 | lan4
----- | --- | ---- | ---- | ---- | ----
lan     |  |  X | X |  X
dtd    |  | | | | X
wan   | X

To avoid complications I wouldnt suggest messing with vlans here, but there is already a WAN VLAN setting elsewhere anyway.

cTurtle98, 2023-02-06:
> To avoid complications I wouldnt suggest messing with vlans here, but there is already a WAN VLAN setting elsewhere anyway.

how would we specify if wan would be tagged or untagged on each interface?

would you build logic in?
psuedocode:
```
if (lan = enabled):
    wan = tagged 
else:
    wan = untagged
```
something like that?

aanon4, 2023-02-06:
So there's an advance setting where you can set the VLAN id of the WAN. I'd assumed a blank setting simply means untagged. As for LAN and WAN on the same port, I'd disallow that unless you'd set the WAN VLAN.

I expect pushback on this feature and I'm not proposing it makes it into the next release anyway. I'd rather it was simple than this, but I'm not sure how to achieve that and still provide what people want.

cTurtle98, 2023-02-06:
an alternate idea

dont have a dedicated d2d port and dedicated lan ports,
instead, all (not wan) ports have tagged vlan2 and un-tagged lan

the wan vlan selected in the other advanced menu option is also enabled on all (not wan) ports

this new option will just be a drop down menu to select which port is the un-tagged wan port
the options list need to include a none option (all the ports are tagged wan, tagged d2d, and un-tagged lan)

aanon4, 2023-02-06:
> There is a confusing double expectation with aredn where it doesn't support double nat because you can't have the upstream network be inside 10.0.0.0/8 or 172.16.0.0/10 but also having a wan vlan instead of routing wan through the d2d network requires that there be double nat
> 
> This could be solved by eliminating the wan vlan entirely and implementing sharing wan through d2d as an option just how there is share wan to lan and share wan to mesh
> 
> It would require every tower site have a device with a dedicated wan port to be the gateway for that d2d network but I don't think that is too bad of a trade off
> 
> My suggestion in my last comment might also require enabling dhcp guarding on those devices to keep d2d devices Lans from interfering with their own Lans (probably why the dedicated ports were done in the first place)

We dont get to take features away. Plenty of folk use the WAN access for their tunnel servers and clients and it can't just go away with an update or be changed substantially so their stuff doesn't work anymore.

The notion of a shared WAN sounds quite different from the original discussion of how we might provide flexibility in how ports are assigned. Doesnt mean its not an interesting idea (do you have a prototype?) but it's probably not what the ticket creator was after.

AJ6GZ, 2023-02-06:
Remember, having all ports LAN untagged would result in DHCP conflicts when
a 2nd/3rd+ node is connected.

On Sun, Feb 5, 2023 at 5:23 PM Ciaran Farley ***@***.***>
wrote:

> an alternate idea
>
> dont have a dedicated d2d port and dedicated lan ports,
> instead, all (not wan) ports have tagged vlan2 and un-tagged lan
> Message ID: ***@***.***>

aanon4, 2023-02-06:
As a way of experimenting here, perhaps the path forward is for someone to write an external package to demo a potential configuration UI? It would need to create the appropriate config files (support exists in nightly) based on the UI changes? Would be easier to get some feedback and ultimately buy-in.

cTurtle98, 2023-02-06:
> The notion of a shared WAN sounds quite different from the original discussion of how we might provide flexibility in how ports are assigned. Doesnt mean its not an interesting idea (do you have a prototype?) but it's probably not what the ticket creator was after.

I have another ticket discussing this #466

mathisono, 2023-02-06:
A few notes... 

@aanon4  I Like the idea! consider adding another column for "tagged ID" #'s. That places all the config in one table. Keep all this stuff collocated in advanced configuration to minimize confusion.

Consider a flag! when the mesh LAN traffic is changed to anything. ( LAN will only be reachable over a tagged connection. )

@AJ6GZ  in regards to "Remember, having all ports LAN untagged would result in DHCP conflicts when
a 2nd/3rd+ node is connected." 

Q; do nodes still check for competing DHCP servers? even if not...

Not a problem! This type of HUB config is intended to be done with a managed switch, not a unmanaged switch. A solution would be to assign an arbitrary vlan id example! being 155, untagged to the port Node 1 is connected too. Then assign another vlan id 156 untagged to Node 2 port, and so on for node 3. This segments each node's Mesh lan to the specific vlan id, that can now be assigned to a port as needed, or trunked along with other tagged vlan ids. 

@cTurtle98 I use tagged wan's to assign my nodes with at hub to a VLAN, that's then exposesd to the ISP, untagged on another port.

This is how I have my AC2 in use. 
https://user-images.githubusercontent.com/618284/209456950-27412b78-bbe3-4ab6-b500-6c72858cf484.PNG

mathisono, 2023-02-06:
This is how I really need a node configured... 
[image]

cTurtle98, 2023-02-06:
> This is how I really need a node configured... [image]

Yeah so it looks like my idea doesn't make sense for that.

> As a way of experimenting here, perhaps the path forward is for someone to write an external package to demo a potential configuration UI? It would need to create the appropriate config files [...]

Is there an example package I can look at to see how to do that?

I want to put together a package for the table idea

aanon4, 2023-02-06:
Okay - this is my stab at a simple utility to configure the AC2 or AC3.

https://github.com/kn6plv/port-config

Feel free to use it as a template to build something better (or just submit PRs to me).  Needs a current nightly to work.

ae6xe, 2023-02-07:
Sorry, I've not had time to soak all the dialog in this thread, but some statements caught my eye.   Just in case this is helpful:

1) joining node LAN's together is done.  Use case is daisy chaining Nanostations -- POE feeds 2nd NS -- NS-1 secondary -> NS-2 main Ethernet ports.   In this case, when the 2nd device boots, DHCP sees an existing service, and does not come up.    LAN devices on both nodes, or if there are 3+ daisy-chained nodes, end up obtaining an IP address from the 1st node to power up.   This makes it a bit more complicated to deal with and all users understanding what is going on. 

2) combining untagged and tagged on the same port issue. When the device has an onboard switch chip, the drivers fail to support both untagged and tagged on the same external switch port.  This has never worked in openwrt releases.  If attempting this use case, be sure to use both tagged and untagged at the same time to reproduce the failure.   (Note, as I recall, the WAN port on the hap ac lite is eth1 and only the other 4 ports are on a switch on eth0--have this issue.)

3) 172.x.x.x issue.   Note, the routing across the AREDN mesh is all based on 10.x.x.x addresses.   There's no conflict when a) the node does not have a tunnel; b) has a local WAN at 172.x.y.z.    If a node has a tunnel and a WAN with 172.x.y.z, it should be uncommon to have a conflict -- possibly acceptable chance of a collision.   The home routers are RFC unrouteable 172.x.y.z/24 subnet, and AREDN a smaller subnet.  Both the home router and the tunnels would have to have the same value for 'x.y' to be in conflict.  If we compare (don't have in front of me to compare for the moment), AREDN should not be using a common RFC unrouteable 'x.y' used in home routers.

cTurtle98, 2023-02-07:
> 1. joining node LAN's together is done.  Use case is daisy chaining Nanostations -- POE feeds 2nd NS -- NS-1 secondary -> NS-2 main Ethernet ports.   In this case, when the 2nd device boots, DHCP [...]

Can you daisy chain hap ac lite?
Port 1 would have to gave d2d enabled on it

aanon4, 2023-02-07:
Just to note that this ticket is about the hAP ac2 and not the hAP lite. Pedantic I know, but the switch is managed very differently on the ac2 and ac3 (using the Distributed Switch Architecture - https://www.kernel.org/doc/html/latest/networking/dsa/dsa.html) compared to the lite.

mathisono, 2023-02-10:
plug-in works! but...

When the WAN VLAN number set (in my case 1) the plug in doesn't recognize the vlan # for the WAN. It still thinks WAN is untagged.  

[image]
[image]

mathisono, 2023-02-10:
If I can use VLAN id #1 like another node do? Should I start using another vlan id at the hub?

aanon4, 2023-02-10:
Type the VLAN you want in the white "Untagged" box in the tool (the tool doesnt interact with advanced config). VLAN must be > 3 (more a limitation of the tool than anything else).

mathisono, 2023-02-10:
I never much liked vlan 1.

aanon4, 2024-12-04:
I think this is done now.

closed 2024-12-04 by aanon4

mathisono, 2024-12-06:
Yes