aredn/aredn #589: Iperfspeed 0.6 Lua not displaying available nodes in CServer or Client list with Nightly 2017+

k1ky, 2022-12-25:
Iperfspeed 0.6 Lua doesn't display any available nodes in the Server or Client list with the new OpenWRT N-2017+ Nightly upgrades.

aanon4, 2022-12-25:
Please try this version: https://github.com/kn6plv/iperfspeed/raw/master/iperfspeed_0.7_all.ipk

k1ky, 2022-12-25:
Looks like it works with 2023 on a GLUSB-150.  Is this version downward compatible if installed on older Firmware versions?

aanon4, 2022-12-25:
Yes.

k1ky, 2022-12-25:
Excellent - You da man!

aanon4, 2022-12-25:
Going to close this one then - thanks!

k1ky, 2022-12-25:
Having issues with Server side talking to clients.  It appears that it works in reverse.

aanon4, 2022-12-25:
The only change to make it work was deleting a library reference which wasn't being used - not sure how it can have reversed

k1ky, 2022-12-25:
I think it needs more work.  See notes in issue.

aanon4, 2022-12-25:
See note above - nothing has changed about how the program runs - only a library reference which isn't being used was removed. I've reopened for now, but dont see how it can be working in reverse compared to before.

kc2cbd, 2022-12-25:
On my GLUSB150 it can't be a server but works fine as a client.

k1ky, 2022-12-25:
And now they are all working on my GLUSB150 and AR750.  Let's let this one soak for a little while and revisit.

k1ky, 2022-12-26:
No further issues with 0.7 at this time. Closing this issue.

closed 2022-12-26 by k1ky