aredn/aredn #590: MeshChat 2.2 doesn't communicate - new OpenWRT Nightlies 2017+

k1ky, 2022-12-25:
MeshChat 2.2 does not see or communicate with any stations in Zone when installed on new OpenWRT Nightly 2017+ updates

aanon4, 2022-12-25:
Please try - https://github.com/kn6plv/meshchat/raw/master/meshchat-api_2.3_all.ipk

k1ky, 2022-12-25:
2.3 Doesn't seem to pick up on the Correct Zone as set up - continuing to test.

k1ky, 2022-12-25:
Attempts to save changes in the Port Forwarding area returns a Bad Gateway message.

aanon4, 2022-12-25:
That ones separate from MeshChat :-(
Looks like there's still more to do on MeshChat as well.

aanon4, 2022-12-25:
Try 2.4

k1ky, 2022-12-25:
Version 2.4 appears to be working!

k1ky, 2022-12-26:
I may still be experiencing some issues with 2.4 Syncing with Zone.

aanon4, 2022-12-26:
Can you attach the support tool data dump?

k1ky, 2022-12-26:
Done - this appears to be a one-way outgoing issue from the new Openwrt firmware.  I installed 2.4 MeshChat on a TPLink CPE510 and it is working both ways and sees all of the stations within the Zone.

k1ky, 2022-12-26:
Looks like MeshChat v2.5 took care of it.  Compatability back to units running v2.2 has been tested. Installed on GLINET AR-750 running N-2031.

k1ky, 2022-12-26:
No further issues with version 2.5 - closing this issue.

closed 2022-12-26 by k1ky