aredn/aredn #606: "Signature check failed" when refreshed on nighty.

mathisono, 2022-12-29:
I was just going to look and see if nano "text editor" was included in the package install repos? When I refreshed the following was returned...

Downloading
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
ges.gz
Updated list of available packages in /var/opkg-lists/aredn_core
Downloading
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
ges.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_base
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag
es.gz
Updated list of available packages in /var/opkg-lists/aredn_arednpackages

Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag
es.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_luci
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.gz

Updated list of available packages in /var/opkg-lists/aredn_packages
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.si
g
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.gz

Updated list of available packages in /var/opkg-lists/aredn_routing
Downloading 
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.sig

Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.g
z
Updated list of available packages in /var/opkg-lists/aredn_telephony
Downloading
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.s
ig
Signature check failed.
Remove wrong Signature file.

dman776, 2022-12-29:
You have to use the latest nightly buildsThanks,Darryl

http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.gz
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.sig
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.gz
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.sig
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.gz
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.si
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.gz
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.sig
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.g
http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.s

mathisono, 2022-12-30:
Thanks! I went ahead and used vi... not my favorite ! 

Downloading http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa ges.gz Updated list of available packages in /var/opkg-lists/aredn_core Downloading http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa ges.sig Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.gz Updated list of available packages in /var/opkg-lists/aredn_base Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/base/Packages.sig Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag es.gz Updated list of available packages in /var/opkg-lists/aredn_arednpackages  Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/arednpackages/Packag es.sig Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.gz Updated list of available packages in /var/opkg-lists/aredn_luci Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/luci/Packages.sig Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.gz  Updated list of available packages in /var/opkg-lists/aredn_packages Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/packages/Packages.si g Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.gz  Updated list of available packages in /var/opkg-lists/aredn_routing Downloading  http://downloads.arednmesh.org/snapshots/packages/mips_24kc/routing/Packages.sig  Signature check passed. Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.g z Updated list of available packages in /var/opkg-lists/aredn_telephony Downloading http://downloads.arednmesh.org/snapshots/packages/mips_24kc/telephony/Packages.s ig Signature check passed.
--

closed 2022-12-30 by mathisono