aredn/aredn #607: GL-iNet AR300M16 bricked or a bug? (aredn-3.22.12.0 or later firmware)

convolved-KY4G, 2022-12-29:
There may be some additional testing needed with GL-iNet AR300M16 routers.  I don’t know to what extent they were able to be tested, but I’ve detailed the problems I’ve been having below.  I don’t know if this is a bug or if I’ve somehow bricked my router, but it is curious that I am able to upload firmware yet still have the problem.
I have a AR300M16 that I had been using successfully with aredn-3.22.8.0-ar71xx-generic-gl-ar300m-squashfs-sysupgrade.bin for quite a while.  I tried updating to aredn-3.22.12.0-ar71xx-generic-gl-ar300m-squashfs-sysupgrade.bin right after it came out and the box has been having trouble ever since with roughly the same behavior regardless of what firmware I put on it (including two recent nightly builds).  I am using the "Using Uboot to Debrick Your Router" procedure found here [https://docs.gl-inet.com/en/3/tutorials/debrick/](url) to upload the firmware.  I upload firmware that I downloaded from the AREDN site, the router reboots and I get the NOCALL page that states that the node is not yet configured.  I go to the setup page.  At this point if I reboot, the router will come back to the same NOCALL page just fine with no issues (Note that the AREDN page is served from 192.168.1.1 at this point still).  If I change anything, even just the password, and click save changes the router reboots and gives the following behavior.  The router will reboot with lights blinking.  It will serve a webpage (I know that (http://10.186.170.153/cgi-bin/setup)/cgi-bin/setup works) for about 2 or 3 seconds and then the router crashes and reboots.  It does this over and over.  Within that very short timeframe, I have sometimes been able to make a quick change and save it before it reboots. At this point it isn't even providing DHCP services. I hardcoded my computer to 10.186.170.154 and was able to navigate to the link above.  I disabled Mesh RF, LAN, LAN Access Point, WAN, and WAN Wifi Client hoping to keep it from crashing, but I've had no luck.  
The nightly builds I have tried are aredn-2024-fa898cc-ath79-glinet_gl-ar300m16-sysupgrade.bin and aredn-2065-599819c-ath79-generic-glinet_gl-ar300m16-squashfs-sysupgrade.bin but they have the same problem.  Rolling back to aredn-3.22.8 did not fix the issue.  It still does the same thing.
If there are any suggestions on what to try next on my end, I would appreciate those as well.  I have screenshots if they are helpful.  Also let me know if this should be on the forum instead of being reported as an issue. My apologies if that is the case.

ab7pa, 2022-12-31:
You should be able to use the debrick procedure you mentioned to install the latest supported release of the [GL.iNet firmware](https://dl.gl-inet.com/?model=ar300m16).  Once back in its original state, you should be able to use the [regular AREDN install procedure](https://docs.arednmesh.org/en/latest/arednGettingStarted/installing_firmware.html#gl-inet-first-install-process) to load AREDN firmware.

convolved-KY4G, 2022-12-31:
Oddly, while I can load AREDN firmware to it and get it to come up to the initial configuration page (which lists the firmware installed), I cannot get it to take any of the recent GL.iNet firmware loads.  It appears to take the firmware, but after it reboots I have found no way to access the device and the only IP address I retrieve is 169.254.202.154.  Even if I try to manually set the IP to 192.168.1.2 or 192.168.8.2, I cannot access the router.

aanon4, 2023-01-17:
Did you make any progress on this? I have an AR300M16 as part of my test farm and I've had no issue with it so far.

convolved-KY4G, 2023-01-17:
Thank you for reporting that you have had no issues in your testing.  No matter what firmware I load, the best I end up with is a very crippled node.  Something has gone wrong with it that cannot be fixed.  At this point, it is just easier to replace it.  I will close out the ticket.

closed 2023-01-17 by convolved-KY4G