aredn/aredn #611: NanoBeam AC Nightly 2065 Firmware Check Failed

cTurtle98, 2022-12-31:
I have nanobeam AC 
NBE-5AC-19 

I held the reset button till I get the 1-3 2-4 blink lights
```
tftp 192.168.1.20
bin
trace on
put xxxx.bin
```

I tried `aredn-2065-599819c-ath79-generic-ubnt_nanobeam-ac-squashfs-factory` but that said error 2 frmware check failed

so then I tried all of these

* aredn-2065-599819c-ath79-generic-ubnt_nanobeam-ac-initramfs-kernel
* aredn-2065-599819c-ath79-generic-ubnt_nanobeam-ac-gen2-initramfs-kernel
* aredn-2065-599819c-ath79-generic-ubnt_nanobeam-ac-gen2-squashfs-factory

all of them said firmware check failed.

dman776, 2022-12-31:
Did you see the new install instructions for Ubiquiti AC devices in the docs?

aanon4, 2022-12-31:
Yes, please use the new method which is documented on the website. You cannot install firmware this way with AC devices as the firmware is not signed.

cTurtle98, 2022-12-31:
I did not...
reading this now
https://docs.arednmesh.org/en/latest/arednGettingStarted/installing_firmware.html#ubiquiti-802-11ac-first-install-process

cTurtle98, 2022-12-31:
what is the initramfs file for? its not used in the instructions

cTurtle98, 2022-12-31:
```
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~/Downloads$ scp aredn-2065-599819c-ath79-generic-ubnt_nanobeam-ac-squashfs-factory.bin ubnt@192.168.1.20:/tmp/factory.bin
ubnt@192.168.1.20's password: 
aredn-2065-599819c-ath79-generic-ubnt 100% 5441KB   3.7MB/s   00:01    
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~/Downloads$ ssh ubnt@192.168.1.20
ubnt@192.168.1.20's password: 
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: NanoBeam 5AC 19
  \ \  \\\  \	MAC: 44D9E7485C87
   \ \_______\
    \|_______|  VERSION: XC.v8.7.0

XC# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdu
mp -R > /tmp/fwupdate.real

XC# 
XC# chmod +x /tmp/fwupdate.real
XC# /tmp/fwupdate.real -m /tmp/factory.bin
Invalid version 'WA.ar934x.v8.5.3-42.AREDN-r19803-9a599fee93'
XC# 
```

flash unsuccessful

Orv, 2023-01-06:
That means that your command to kill the signature validation of fwupdate failed.  Try it again.

Orv W6BI

cTurtle98, 2023-01-11:
> That means that your command to kill the signature validation of fwupdate failed. Try it again.
> 
> Orv W6BI

I did this and I am still getting the same thing.

how do I get the XC firmware virsion?

my prompt says `XC` and the invalid version message starts with `WA`

aanon4, 2023-01-11:
I'm guessing this isnt the Gen2 model (which I've checked)? We've not validate the non Gen2 model so I'm not sure what to expect here without access to the hardware.

Orv, 2023-01-12:
This procedure outlines how to retrieve the correct AirOS software: 

https://docs.arednmesh.org/en/latest/_downloads/cc242dc167cc98af688b0c1245d47236/Ubiquiti_AC_First_Install_Checklist.pdf

cTurtle98, 2023-01-12:
> I'm guessing this isnt the Gen2 model (which I've checked)? We've not validate the non Gen2 model so I'm not sure what to expect here without access to the hardware.

yes this is a gen1

how far are you from santa cruz?
would if be viable for you to borrow mine?

> This procedure outlines how to retrieve and install the correct AirOS software:
> 
> https://docs.arednmesh.org/en/latest/_downloads/cc242dc167cc98af688b0c1245d47236/Ubiquiti_AC_First_Install_Checklist.pdf

I am following that guide when I do this. 
I downloaded 8.7.0 from the links in the guide and tftp put both of them but my radio took the xc version.
so my radio is a gen 1 xc

cTurtle98, 2023-01-12:
> This procedure outlines how to retrieve and install the correct AirOS software:
> 
> https://docs.arednmesh.org/en/latest/_downloads/cc242dc167cc98af688b0c1245d47236/Ubiquiti_AC_First_Install_Checklist.pdf

how important is it that the username and password are whats stated in the guide?
i used a different one to do this.
im going to try with the same user and password as the guide now

cTurtle98, 2023-01-12:
> how important is it that the username and password are whats stated in the guide? i used a different one to do this. im going to try with the same user and password as the guide now

I tried again with the same password as the docs and got the same result

```
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~$ ssh admin@192.168.1.20
admin@192.168.1.20's password: 
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: NanoBeam 5AC 19
  \ \  \\\  \	MAC: 802AA824387A
   \ \_______\
    \|_______|  VERSION: XC.v8.7.0

XC# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R > /tmp/fwupdate.real
XC# chmod +x /tmp/fwupdate.real
XC# /tmp/fwupdate.real -m /tmp/factory.bin
Invalid version 'WA.ar934x.v8.5.3-42.AREDN-r19803-9a599fee93'
XC# 
```

cTurtle98, 2023-01-13:
> Not sure what's going on here except to think that we need a different "patch" for fwupdate.real on this device. Can you log into the device and send the output of this:
> 
> `hexdump -Cv /bin/ubntbox | grep '14 40 fe 27'`

```
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~$ ssh admin@192.168.1.20
admin@192.168.1.20's password: 
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: NanoBeam 5AC 19
  \ \  \\\  \	MAC: 802AA824387A
   \ \_______\
    \|_______|  VERSION: XC.v8.7.0

XC# 
XC# hexdump -Cv /bin/ubntbox | grep '14 40 fe 27'
00009ca0  26 a6 fe f8 14 40 fe 27  af a2 02 fc 82 02 00 03  |&....@.'........|
XC# 
XC# 
XC# 
XC# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R > /tmp/fwupdate.real
XC# 
XC# hexdump -Cv /bin/ubntbox | grep '14 40 fe 27'
00009ca0  26 a6 fe f8 14 40 fe 27  af a2 02 fc 82 02 00 03  |&....@.'........|
XC# 
XC# 
```

aanon4, 2023-01-17:
Looking through some documentation this evening I noticed there is a different build - ubnt_nanobeam-ac-xc - which might be what we need for this device. We've not currently building it as a nightly but I can build a custom one for you to test? If it works we can then add it.

cTurtle98, 2023-01-17:
> Looking through some documentation this evening I noticed there is a different build - ubnt_nanobeam-ac-xc - which might be what we need for this device. We've not currently building it as a nightly but I can build a custom one for you to test? If it works we can then add it.

Sure. I'd love to test it. That must be why the firmware it's trying to install is wa but my model is an xc

aanon4, 2023-01-17:
I sent a build to the email registered with your sfarc slack account. Let me know if you need it sent elsewhere.

cTurtle98, 2023-01-17:
```
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~/Downloads$ scp aredn-kn6plv-main-3366f9b-ath79-generic-ubnt_nanobeam-ac-xc-squashfs-factory.bin  admin@192.168.1.20:/tmp/factory.bin
admin@192.168.1.20's password: 
Permission denied, please try again.
admin@192.168.1.20's password: 
aredn-kn6plv-main-3366f9b-ath79-generic-ubnt_ 100% 5441KB   3.7MB/s   00:01    
ciaran@ciaran-ThinkPad-P14s-Gen-2a:~/Downloads$ ssh admin@192.168.1.20
admin@192.168.1.20's password: 
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: NanoBeam 5AC 19
  \ \  \\\  \	MAC: 802AA824387A
   \ \_______\
    \|_______|  VERSION: XC.v8.7.0

XC# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R > 
/tmp/fwupdate.real

XC# 
XC# chmod +x /tmp/fwupdate.real
XC# /tmp/fwupdate.real -m /tmp/factory.bin

Found  Active on[2] ...
Storing Active[2] ... [%100]
Writing 'kernel         ' to /dev/mtd2(kernel         ) ...  [%100]
Writing 'rootfs         ' to /dev/mtd3(rootfs         ) ...  [%100]
Done

```
the first time setup page is different than i'm used to
[image]

sysinfo page says:
```
 node: NOCALL-36-56-122
model: 0xe4f5

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe4f5
Device HAS BEEN TESTED AS UNSUPPORTED

br-lan 80:2A:A8:25:38:7A
eth0   80:2A:A8:25:38:7A
eth0.1 80:2A:A8:25:38:7A
eth0.2 80:2A:A8:25:38:7A
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  80:2A:A8:24:38:7A

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9558 ver 1 rev 0
machine			: Ubiquiti NanoBeam AC Gen1 (XC)
processor		: 0
cpu model		: MIPS 74Kc V5.0
BogoMIPS		: 358.80
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit cdmm contextconfig perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='80:2a:a8:24:38:7a'
hsmmmesh.settings.mac2='36.56.122'
hsmmmesh.settings.node='NOCALL-36-56-122'
hsmmmesh.settings.dtdmac='37.56.122'
```

here is the support data

[supportdata-NOCALL-202301030028.gz](https://github.com/aredn/aredn/files/10439117/supportdata-NOCALL-202301030028.gz)

when I try to change the host name and password for the first time setup I get a bad gateway error

aanon4, 2023-01-17:
Thanks. I'll add it to the nightly.

aanon4, 2023-01-17:
https://github.com/aredn/aredn/pull/655

aanon4, 2023-01-17:
Can you follow the instruction in the PR written by @ae6xe to extract the missing power data? Sorry.

aanon4, 2023-01-18:
Fantastic! Thanks.

cTurtle98, 2023-01-18:
should I try last night's nighty on my radios or wait till tomorrow when the power offset is implemented?

aanon4, 2023-01-18:
The power offset is a relatively minor thing. If you'd try today's nightly it should fix everything else (hopefully)

cTurtle98, 2023-01-18:
[image]

looks like its working

here is the support data for good measure
[supportdata-K0CDF-NBAC-72-92-135-202301030032.gz](https://github.com/aredn/aredn/files/10450922/supportdata-K0CDF-NBAC-72-92-135-202301030032.gz)

cTurtle98, 2023-01-18:
when I do the first time setup save it doesnt come back up properly. I have to wait till its back to "on" lights then unplug it and plug it back in

aanon4, 2023-01-19:
Sorry - I'll need more detail as I dont understand what you mean by this.

cTurtle98, 2023-01-19:
> Sorry - I'll need more detail as I dont understand what you mean by this.

By first time setup I mean going to 192.168.1.1 and setting the hostname and password and clicking save

It didn't come back up after that. I had to unplug it and plug I back in. 

After hitting save the dhcp server for 192.168.1.x wasn't there anymore but the 10.x.x.x never came up. My laptop kept giving itself zeroconf addresses

aanon4, 2023-01-20:
A couple of things:
1. A normal reboot works fine? This was just that one time?
2. Assuming 'yes' to the above, do you want to repeat the experiment to see if its a consistent thing or just happened this one time? If you do repeat it, can you give it at least 30 minutes (extremely excessive!) to come back, and if it doesnt, try unplugged and replugging the ethernet to the laptop (so dont unpower it).
3. If this is consistent, please open a new issue for this new problem so we can close this one.

cTurtle98, 2023-01-20:
I have two nanobeam ac 5 devices that I flashed and they both did it

I will test again and open a new issue if its persistent 

Thank you for building support for the xc variant

cTurtle98, 2023-01-20:
I forgot to answer your question

> 1. A normal reboot works fine? This was just that one time?

I changed the time zone and hit the save button then the reboot button and it came back up without removing power

So I think that is a yes

I have not tried just using the reboot button without having saved settings changes first

closed 2023-01-20 by cTurtle98