aredn/aredn #619: mikrotik rb-912uag-5hpnd firmware file not valid

kc2cbd, 2023-01-05:
Trying to upgrade a node to 2096 and also tried 2024.
Node says 
Firmware CANNOT be updated
firmware file is not valid

[image]

aanon4, 2023-01-05:
Could you log into the node and send me the output when you run the following:

```for i in /tmp/sysinfo/* ; do echo $i ; cat $i; do```

Thanks

aanon4, 2023-01-05:
Also, please confirm the specific firmware filename you were loading.

kc2cbd, 2023-01-05:
Firmware that I was using was aredn-2096-fce9629-ath79-mikrotik-mikrotik_routerboard-912uag-5hpnd-squashfs-sysupgrade.bin
[image]

kc2cbd, 2023-01-08:
If we were able to get the newer firmware on this device, would it upgrade fine after that?

aanon4, 2023-01-08:
Yes - it's just the transition which is the limitation.

aanon4, 2023-02-20:
Closing as an upgrade path from 3.22.12.0 to the next release will not be possible. A re-install must be done and then upgrades can continue.

closed 2023-02-20 by aanon4