aredn/aredn #625: Mikrotik routerboard-952ui-5ac2nd

kc2cbd, 2023-01-07:
The ability for the hAP to go back to older firmware is now broken.
Firmware CANNOT be updated
firmware file is not valid: platform check image failed
using the firmware aredn-2089-bc77ff8-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-squashfs-sysupgrade.bin

aanon4, 2023-01-07:
Can you confirm which firmware you running and which you are trying to revert to?

kc2cbd, 2023-01-07:
Currently on aredn-2103-5214d35-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-squashfs-sysupgrade.bin

kc2cbd, 2023-01-07:
The USB150 won't go back to and older firmware either.

aanon4, 2023-01-07:
https://github.com/aredn/aredn/pull/626

aanon4, 2023-01-07:
The fix for the moment:
Log into your node:
type:
```echo "#!/bin/true" > /usr/local/bin/firmwarecheck.sh```

kc2cbd, 2023-01-07:
That allowed my nodes to go back to an older firmware

closed 2023-01-08 by kc2cbd